b-rad-NDi/Ubuntu-media-tree-kernel-builder #136: HVR-2250 & 2255 card firmware not recognized

techiegirl2day, 2022-01-25:
These are the cards I have installed on MX Linux 21 ( debian 11 bullseye) tried with both kernel 5.10 & 4.19 

06:00.0 Multimedia video controller: Conexant Systems, Inc. CX23887/8 PCIe Broadcast Audio and Video Decoder with 3D Comb (rev 04)
07:00.0 Multimedia video controller: Conexant Systems, Inc. CX23887/8 PCIe Broadcast Audio and Video Decoder with 3D Comb (rev 04)

09:00.0 Multimedia controller: Philips Semiconductors SAA7164 (rev 81)  model 2255

**0b:00.0 Multimedia controller: Philips Semiconductors SAA7164 (rev 81)  model 2250
   this card I am unable to link to software or tune it in linux.  All seem to function correctly in win 10

dmesg error related to these cards  I cut & paste 
NXP7164-2010-03-10.1.fw
v4l-saa7164-1.0.3-3.fw
into /lib/firmware since they are not included in my kernels.

[    8.479902] saa7164[0]: Your board isn't known (yet) to the driver.
                      saa7164[0]: Try to pick one of the existing card configs via
                      saa7164[0]: card=<n> insmod option.  Updating to the latest
                      saa7164[0]: version might help as well.
[    8.479988] saa7164[0]: Here are valid choices for the card=<n> insmod option:
[    8.480066] saa7164[0]:    card=0 -> Unknown
[    8.480112] saa7164[0]:    card=1 -> Generic Rev2
[    8.480159] saa7164[0]:    card=2 -> Generic Rev3
[    8.480205] saa7164[0]:    card=3 -> Hauppauge WinTV-HVR2250
[    8.480253] saa7164[0]:    card=4 -> Hauppauge WinTV-HVR2200
[    8.480301] saa7164[0]:    card=5 -> Hauppauge WinTV-HVR2200
[    8.480349] saa7164[0]:    card=6 -> Hauppauge WinTV-HVR2200
[    8.480397] saa7164[0]:    card=7 -> Hauppauge WinTV-HVR2250
[    8.480444] saa7164[0]:    card=8 -> Hauppauge WinTV-HVR2250
[    8.480492] saa7164[0]:    card=9 -> Hauppauge WinTV-HVR2200
[    8.480540] saa7164[0]:    card=10 -> Hauppauge WinTV-HVR2200
[    8.480588] saa7164[0]:    card=11 -> Hauppauge WinTV-HVR2255(proto)
[    8.480637] saa7164[0]:    card=12 -> Hauppauge WinTV-HVR2255
[    8.480685] saa7164[0]:    card=13 -> Hauppauge WinTV-HVR2205
:
:
starting firmware download(2)
[   19.068098] saa7164_downloadimage() Image downloaded, booting...
[   20.624078] saa7164_downloadimage() Image booted successfully.
[   20.624093] firmware download complete.
[   20.676475] tveeprom: Hauppauge model 88061, rev C4F2, serial# XXXXXXXX
[   20.676477] tveeprom: MAC address is XXXXXXXX
[   20.676479] tveeprom: tuner model is NXP 18271C2_716x (idx 152, type 4)
[   20.676481] tveeprom: TV standards NTSC(M) ATSC/DVB Digital (eeprom 0x88)
[   20.676483] tveeprom: audio processor is SAA7164 (idx 43)
[   20.676484] tveeprom: decoder processor is SAA7164 (idx 40)
[   20.676485] tveeprom: has radio, has IR receiver, has no IR transmitter
[   20.676487] saa7164[1]: Hauppauge eeprom: model=88061
[   21.026940] tda18271 7-0060: creating new instance
[   21.031738] tda18271: TDA18271HD/C2 detected @ 7-0060
[   21.336047] dvbdev: DVB: registering new adapter (saa7164)
[   21.336053] saa7164 0000:09:00.0: DVB: registering adapter 4 frontend 0 (Samsung S5H1411 QAM/8VSB Frontend)...
[   21.624577] tda18271 8-0060: creating new instance
[   21.628831] tda18271: TDA18271HD/C2 detected @ 8-0060
[   21.928832] tda18271: performing RF tracking filter calibration
[   24.474135] tda18271: RF tracking filter calibration complete
[   24.477432] dvbdev: DVB: registering new adapter (saa7164)
[   24.477437] saa7164 0000:09:00.0: DVB: registering adapter 5 frontend 0 (Samsung S5H1411 QAM/8VSB Frontend)...
[   24.513992] saa7164[1]: registered device video4 [mpeg]
[   24.741321] saa7164[1]: registered device video5 [mpeg]
[   24.953241] saa7164[1]: registered device vbi2 [vbi]
[   24.953410] saa7164[1]: registered device vbi3 [vbi]
[   24.955236] saa7164 driver loaded

What is needed to trigger the proper download & registering of the firmware for these cards?

I had found info online that I needed to add  options doc in /etc/modprobe.d/ with text 
options saa7164 card=8
options saa7164 card=12  

not sure  if this works with more than one card instance. or if I have correctly indentified the correct card model versions it is looking for.

I had hoped that you would have created a more viable option for us debian users. I know you are swamped as a one man band show. I again hope you &/.or your bosses will see the light to funnel more resources into Linux support. 
~                                                                                                   
~                                                                                                   
~                                                                                                   
~                                                                                                   
~                                                                                                   
~                                                                                                   
~                                                                                                   
~                                                                                                   
~                                                                                                   
~                                                                                                   
~                                                                                                   
~                                                                                                   
~                                                                                                   
~                                                                                                   
~                                                                                                   
~                                                                                                   
~                                                                                                   
~                                                                                                   
~                                                                                                   
~                                                                                                   
~                                                                                                   
~                                                                                                   
~                                                                                                   
~                                                                                                   
~                                                                                                   
"/etc/modprobe.d/options" [readonly] 2L, 47B

techiegirl2day, 2022-01-27:
If I understand this correctly, what is needed is an update to the insmod so that the saa7164 card is recognized & acknowledged since the autodetect for the 2255 card fails. I am not that savy with Linux. Please advice how to amend this myself or maybe you will patch this for all of us that use that card?

techiegirl2day, 2022-01-27:
after creating the requested changes new error msg emerged. 

saa7164_api_modify_gpio() error, ret = 0x13
[   19.820937] saa7164_api_modify_gpio() error, ret = 0x13
[   19.853411] saa7164_api_modify_gpio() error, ret = 0x13
[   19.854966] saa7164_api_modify_gpio() error, ret = 0x13
[   19.855825] saa7164_api_i2c_read() error, ret(1) = 0x13
[   19.859696] saa7164_api_i2c_read() error, ret(1) = 0x13
[   19.859702] lgdt3306a: error (addr 59 reg 0000 error (ret == -5)
[   19.859704] lgdt3306a: error -5 on line 1856
[   19.859706] lgdt3306a: unable to detect LGDT3306A hardware
[   19.859712] saa7164_dvb_register() Frontend initialization failed
[   19.859714] saa7164_initdev() Failed to register dvb adapters on porta
[   19.861190] saa7164_api_i2c_read() error, ret(1) = 0x13
[   19.861195] lgdt3306a: error (addr 0e reg 0000 error (ret == -5)
[   19.861197] lgdt3306a: error -5 on line 1856
[   19.861199] lgdt3306a: unable to detect LGDT3306A hardware
[   19.861205] saa7164_dvb_register() Frontend initialization failed
[   19.861207] saa7164_initdev() Failed to register dvb adapters on portb
[   19.861210] saa7164_encoder_register() failed (errno = -19), NO PCI configuration
[   19.861211] saa7164_initdev() Failed to register mpeg encoder
[   19.861213] saa7164_encoder_register() failed (errno = -19), NO PCI configuration
[   19.861215] saa7164_initdev() Failed to register mpeg encoder
[   19.861217] saa7164_vbi_register() failed (errno = -19), NO PCI configuration
[   19.861219] saa7164_initdev() Failed to register vbi device
[   19.861221] saa7164_vbi_register() failed (errno = -19), NO PCI configuration
[   19.861222] saa7164_initdev() Failed to register vbi device

I am not sure if this is because my entry
options  saa7164 card=12 was wrong card match or something else?

Please consider a dkms version so it will be easier for those of us running debian rather than ubuntu can address firmware & other video card config in a trouble free manner.

shspvr, 2022-01-27:
> 
> Please consider a dkms version so it will be easier for those of us running debian rather than ubuntu can address firmware & other video card config in a trouble free manner.

I should point out to you that Ubuntu is based on Debian just like MX Mint
Have you try add v4l-saa7164-1.0.2-3.fw

techiegirl2day, 2022-01-28:
Mx Linux is not mint or ubuntu. It is secure debian based on sysinit boot not ubuntu. How would an older driver help?

b-rad-NDi, 2022-01-28:
DKMS would not be the way to debug anything. Compiling and testing a mainline kernel would be.

I'll ask support if they can give me any info about 2250 rev81

b-rad-NDi, 2022-01-28:
Also, are you even running my media tree kernel? This repo is pretty much specifically about those kernel builds. If you are looking for support for something outside of my builds then you should file a ticket on my media tree repo to make it clearer you need generic Linux kernel support.

techiegirl2day, 2022-01-28:
The card error message above  is from a 2255.  I changed to another 2250  The common denominator with the affected cards are the  LG Digital electronics LGDT3306a components. My other detected 2250/2255 generation card uses Samsung components.

$ dmesg | grep saa7164
[   48.820405] saa7164[0]: Your board isn't known (yet) to the driver.
               saa7164[0]: Try to pick one of the existing card configs via
               saa7164[0]: card=<n> insmod option.  Updating to the latest
               saa7164[0]: version might help as well.
[   48.820410] saa7164[0]: Here are valid choices for the card=<n> insmod option:
[   48.820411] saa7164[0]:    card=0 -> Unknown
[   48.820412] saa7164[0]:    card=1 -> Generic Rev2
[   48.820413] saa7164[0]:    card=2 -> Generic Rev3
[   48.820414] saa7164[0]:    card=3 -> Hauppauge WinTV-HVR2250
[   48.820415] saa7164[0]:    card=4 -> Hauppauge WinTV-HVR2200
[   48.820416] saa7164[0]:    card=5 -> Hauppauge WinTV-HVR2200
[   48.820417] saa7164[0]:    card=6 -> Hauppauge WinTV-HVR2200
[   48.820418] saa7164[0]:    card=7 -> Hauppauge WinTV-HVR2250
[   48.820419] saa7164[0]:    card=8 -> Hauppauge WinTV-HVR2250
[   48.820420] saa7164[0]:    card=9 -> Hauppauge WinTV-HVR2200
[   48.820421] saa7164[0]:    card=10 -> Hauppauge WinTV-HVR2200
[   48.820422] saa7164[0]:    card=11 -> Hauppauge WinTV-HVR2255(proto)
[   48.820423] saa7164[0]:    card=12 -> Hauppauge WinTV-HVR2255
[   48.820424] saa7164[0]:    card=13 -> Hauppauge WinTV-HVR2205
**_[   48.820458] CORE saa7164[0]: subsystem: 1131:0000, board: Unknown [card=0,autodetected]
[   48.820461] saa7164[0]/0: found at 0000:0b:00.0, rev: 129, irq: 17, latency: 0, mmio: 0xfb800000
[   48.820528] saa7164_initdev() Unsupported board detected, registering without firmware__**
[   48.820616] CORE saa7164[1]: subsystem: 0070:8851, board: Hauppauge WinTV-HVR2250 [card=7,autodetected]
[   48.820619] saa7164[1]/0: found at 0000:09:00.0, rev: 129, irq: 16, latency: 0, mmio: 0xfac00000
[   48.941221] saa7164_downloadfirmware() no first image
[   48.941233] saa7164_downloadfirmware() Waiting for firmware upload (NXP7164-2010-03-10.1.fw)
[   49.103823] saa7164_downloadfirmware() firmware read 4019072 bytes.
[   49.103825] saa7164_downloadfirmware() firmware loaded.
[   49.103828] saa7164_downloadfirmware() SecBootLoader.FileSize = 4019072
[   49.103833] saa7164_downloadfirmware() FirmwareSize = 0x1fd6
[   49.103833] saa7164_downloadfirmware() BSLSize = 0x0
[   49.103834] saa7164_downloadfirmware() Reserved = 0x0
[   49.103834] saa7164_downloadfirmware() Version = 0x1661c00
[   55.964214] saa7164_downloadimage() Image downloaded, booting...
[   56.068190] saa7164_downloadimage() Image booted successfully.
[   58.548227] saa7164_downloadimage() Image downloaded, booting...
[   60.732191] saa7164_downloadimage() Image booted successfully.
[   60.778586] saa7164[1]: Hauppauge eeprom: model=88061
[   61.362967] dvbdev: DVB: registering new adapter (saa7164)
[   61.362973] saa7164 0000:09:00.0: DVB: registering adapter 4 frontend 0 (Samsung S5H1411 QAM/8VSB Frontend)...
[   64.139611] dvbdev: DVB: registering new adapter (saa7164)
[   64.139615] saa7164 0000:09:00.0: DVB: registering adapter 5 frontend 0 (Samsung S5H1411 QAM/8VSB Frontend)...
[   64.175492] saa7164[1]: registered device video2 [mpeg]
[   64.401840] saa7164[1]: registered device video3 [mpeg]
[   64.613716] saa7164[1]: registered device vbi0 [vbi]
[   64.613937] saa7164[1]: registered device vbi1 [vbi]
[   64.615950] saa7164 driver loaded

 Not sure if it is related to kernel or driver specific?  I had taken pc apart to clean & the unknown detection problem surfaced after reboot.  I tested same cards in WIn 10 & they function without issues there. That is my way of weeding out faulty card vs linux code.  I have no experience with building kernels. I would be willing to give it a shot if this will remedy my issues. I noticed from online search that others had issues with saa7164 cards in past The fact that newer Quad card works does not go unnoticed. I don't want to have to buy more newer model cards with different components just to get functionality. If that is the case & will move to network based cards that don't have driver concerns.

b-rad-NDi, 2022-01-28:
Wait a second. The 2250 did work, then you cleaned your pc and it stopped working?

shspvr, 2022-01-28:
> Mx Linux is not mint or ubuntu. It is secure debian based on sysinit boot not ubuntu. How would an older driver help?

Both is built on Debian's architecture

shspvr, 2022-01-28:
> Wait a second. The 2250 did work, then you cleaned your pc and it stopped working?

That is odd too

techiegirl2day, 2022-01-29:
Okay after taking pc apart again & reseating cards & rebooting a couple of times as well as reinstalling drivers & firmware to make sure they weren't corrupted everything is working again. I couldn't get setup to recognize multiple cards of same model.  I am not sure if that was still an outstanding issue. Thanx for feedback.

closed 2022-01-29 by techiegirl2day