b-rad-NDi/Ubuntu-media-tree-kernel-builder #137: USB tuner not working in Plex (Ubuntu 21.10

PSUHammer, 2022-03-25:
Hello,

Sorry for a noob related question but I am having issues getting a USB Xbox One tuner working with a new Ubuntu Plex install.  It works fine in Windows (confirmed today) and has for some time.  I ran through all the steps outlined here and Plex indeed recognizes the Tuner and allows me to start scanning for channels.  But, it comes back empty.   Happy to provide any logs or outputs that may be helpful or if there is a common oversight that causes this behavior, please point me in the right direction.

Plex Server Version#: 1.25.8.5663
OS:  Ubuntu 21.10 x64 deb install
MB:  Asus Prime Z370-A motherboard
Proc:  Intel i7 8700
Tuner Make/Model: Digital TV Tuner for Xbox One (but Plex sees it as WinTV-HVR-955Q)

Thank you in advance for the for any assistance!

PSUHammer, 2022-03-27:
Quick update.  I am troubleshooting my antenna at this point as I was only able to get the tuner to work with a windows laptop closer to the antenna (shorter cable run).  Even then, my signal strength dropped by about 80%.

Please ignore this.  I was mistaken that it was Linux but my Windows machine works sparingly.  I have a feeling the antenna connection is a problem.

shspvr, 2022-03-28:
> Quick update. I am troubleshooting my antenna at this point as I was only able to get the tuner to work with a windows laptop closer to the antenna (shorter cable run). Even then, my signal strength [...]

Yes Digital TV Tuner for Xbox One is nothing more than a WinTV-HVR-955Q with just a diff hardware sub-id

Sound like you could have a junk cable or really old RG59 cable line you should replace it with RG6 Quad Shield Coaxial Cable Solid Copper.
Also if have long cable run you a consider a Preamplifier and also replace Outdoor Balun/Matching Transformer at Antenna head sometime they can go bad and can causes all kind of weirdness.
I hope you have good antenna can post a photo of it or do you know the make and model of it.

PSUHammer, 2022-03-28:
Sure, I have a photo.  I will say my home is 5 years old (so newer) and all cable is RG6.  My run is from my attic two floors down so roughly about 30 to 40 feet, well within distance threshold.

The antenna worked as of the day I installed Ubuntu.  So, it was a massive coincidence that the antenna or cabling just happened to stop working then, but it is possible.  Continuing to troubleshoot and test the lines.

[image]

[image]

shspvr, 2022-03-28:
I see you have a Clearstream 2 Max with reflector assembly
What is that little box with white cable connected to it

PSUHammer, 2022-03-28:
To be honest, I don't know.  Looks like a small amplifier but it isn't powered.  That's what I am focusing on.  Hard to get to this part of my attic so I have some work to do.

shspvr, 2022-03-28:
Then remove it and go direct to the antenna input

Saentist, 2022-03-28:
Post result of DMESG
unpug and plug tuner and execute in terminal
``dmesg | egrep "firmware|dvb|frontend``
post it as text not as picture

Fortuneteller is on vacation, be more detailed.
is it DVB/ATSC signal

What's that stupidity to mount outside antenna inside?
make picture of this linear amplifier on cable before antenna.

Guess under Linux LNA on tuner is not activated by Plex.

PSUHammer, 2022-03-28:
> Post result of DMESG unpug and plug tuner and execute in terminal `dmesg | egrep "firmware|dvb|frontend` post it as text not as picture
>
> Fortuneteller is on vacation, be more detailed. is it [...]

I will assume your comments are a result of a language barrier and not intended to be condescending.   Regardless, my local signal is ATSC.   The antenna is an "indoor/outdoor" antenna and attic installation is very typical and standard.  Especially if you don't want an ugly antenna on your roof and/or you have a homeowners association that doesn't allow it.  I am able to pull 20+ local channels on my Windows machine with the same setup via Plex with most channels over 90% signal strength.

The tuner is listed as supported by Plex and others have used it successfully.

I did as suggested and pulled/reseated the tuner and ran `dmesg | egrep "firmware|dvb|frontend`

No output.

shspvr, 2022-03-28:
> No bueno. Everything is working via Windows using WinTV v10 app. I am showing good signal. Just plugged the tuner device into my win laptop (all other environment variables the same)...boom. 95% signal quality. Validated that cables, signal strength, etc. all check out.
> 
> Plugged into my Ubuntu server, the server sees the adapter, scanning channels via Kaffine or Plex result in zero finds. Definitely Linux/driver problems.
> 
> As another thing to try, I just bought a new tuner, the Hauppauge WinTV-dualHD to rule out any issues with that device. Same problems.
> 
> Are there logs for this driver to show when channels are being scanned? I am somewhat of an Ubuntu noob but if there are specific logs that are generated when a channel scan happens, I am definitely technical enough to pull them up and review.

Did you configure the device set to ATSC

PSUHammer, 2022-03-28:
I choose ATSC in Kaffine when I scan, yes.  I don't believe that is something you set manually in Plex.   That said, I just kicked off a channel scan and ran <sudo dmesg> to see if there were any logs....

```text
[22173.838778] audit: type=1326 audit(1648508920.513:554): auid=1000 uid=1000 gid=1000 ses=2 subj=snap.firefox.firefox pid=15806 comm="MainThread" exe="/snap/firefox/1154/usr/lib/firefox/firefox" sig=0 arch=c000003e syscall=314 compat=0 ip=0x7efe7ae8876d code=0x50000
[22175.762809] audit: type=1326 audit(1648508922.437:555): auid=1000 uid=1000 gid=1000 ses=2 subj=snap.firefox.firefox pid=15850 comm="MainThread" exe="/snap/firefox/1154/usr/lib/firefox/firefox" sig=0 arch=c000003e syscall=314 compat=0 ip=0x7f0881aed76d code=0x50000
[22226.529560] audit: type=1326 audit(1648508973.205:556): auid=1000 uid=1000 gid=1000 ses=2 subj=snap.firefox.firefox pid=7319 comm=49736F6C617465642057656220436F exe="/snap/firefox/1154/usr/lib/firefox/firefox" sig=0 arch=c000003e syscall=312 compat=0 ip=0x7f1398bf076d code=0x50000
[22727.899897] si2157 9-0060: found a 'Silicon Labs Si2157-A30 ROM 0x50'
[22727.899949] si2157 9-0060: Can't continue without a firmware.
[22727.966421] si2157 10-0062: found a 'Silicon Labs Si2157-A30 ROM 0x50'
[22727.966471] si2157 10-0062: Can't continue without a firmware.
[22745.073296] si2157 9-0060: found a 'Silicon Labs Si2157-A30 ROM 0x50'
[22745.073309] si2157 9-0060: Can't continue without a firmware.
[22800.941542] si2157 9-0060: found a 'Silicon Labs Si2157-A30 ROM 0x50'
[22800.941630] si2157 9-0060: Can't continue without a firmware.
[22800.949080] si2157 10-0062: found a 'Silicon Labs Si2157-A30 ROM 0x50'
[22800.949135] si2157 10-0062: Can't continue without a firmware.
```

shspvr, 2022-03-28:
Yes you should always run sudo apt-get install linux-mediatree

PSUHammer, 2022-03-29:
Yeah, the only reason I did not was the blurb at this site:  https://www.hauppauge.com/pages/support/support_linux.html#main

It basically says if you are in the US, you should be good.   Well....nope!

Saentist, 2022-03-29:
> Yeah, the only reason I did not was the blurb at this site: https://www.hauppauge.com/pages/support/support_linux.html#main
> 
> It basically says if you are in the US, you should be good. Well....nope!

http://en.dtvstatus.net/ATSC_DVB-T_ISDB-T_en_archive012007.html
atsc users is tiny percent of world digital tv population and manual exclude it ;)
https://en.wikipedia.org/wiki/Digital_television

command
`` dmesg -T| egrep "firmware|dvb|frontend" `` /sudo if not root/
work perfectly

b-rad-NDi, 2022-04-04:
This is a new "bug" in the kernel. The firmware file should not be required, but a new check forces it. As stated, installing linux-firmware-hauppauge will fix this. Please close if you find that is the case.

PSUHammer, 2022-04-04:
> This is a new "bug" in the kernel. The firmware file should not be required, but a new check forces it. As stated, installing linux-firmware-hauppauge will fix this. Please close if you find that is the case.

Thanks Brad!   I tried that on a whim and it worked.  Consider this closed.

closed 2022-04-04 by b-rad-NDi