b-rad-NDi/Ubuntu-media-tree-kernel-builder #138: linux-hwe-mediatree

micenusa, 2022-03-28:
According to the website:
https://hauppauge.com/pages/support/support_linux.html
I should install the linux-hwe-mediatree version 

the dpkg -l | grep linux-image-generic-hwe response is:
ii  linux-image-generic-hwe-20.04              5.13.0.37.46                               amd64        Generic Linux kernel image

and after I install the PPA, I get:
sudo apt-get install linux-hwe-mediatree
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: **Unable to locate package linux-hwe-mediatree**

micenusa, 2022-03-28:
I'm very new to this linux environment...I assume that I will stick around

PSUHammer, 2022-03-28:
Are you running Ubuntu?   Did you install v21.10?   I ask as I am curious if you are able to get your tuner working in that scenario.

PSUHammer, 2022-03-28:
Where do you see that error?  After running `sudo apt-get install linux-mediatree`??

shspvr, 2022-04-01:
First of all just add PPA
sudo add-apt-repository ppa:b-rad/kernel+mediatree+hauppauge
sudo apt-get update
sudo apt-get install linux-firmware-hauppauge
Then reboot the system
You don't need to install the Install the media tree
This work on Ubuntu 20.04.4 LTS on up

micenusa, 2022-04-05:
The information on the website is still wrong! 
I will let the issue open until they will gave the correct info on the website!
noob people like me need correct and easy to follow instructions in order to deal with this linux stuf!

b-rad-NDi, 2022-04-05:
Paste the full contents of the output from these commands:
```
sudo apt-get update
sudo apt-get install linux-hwe-mediatree
```

micenusa, 2022-04-05:
yeah, thanks..But why there is wrong information on the oficial website?

micenusa, 2024-09-22:
not important

closed 2024-09-22 by micenusa