b-rad-NDi/Ubuntu-media-tree-kernel-builder #139: 461e ts_clk problems

psyborg55, 2022-03-30:
Hi

I try to use several drivers with this device, and the transport stream is corrupted with all of them if the ts clk is set to 24000.
Default linux driver sets it to 16000 and this work fine, but i see reference driver is setting the value of 24000.

Can you check what windows driver does in this case and confirm whether it is hardware limitation or there is an actual bug  in linux drivers?

psyborg55, 2022-04-04:
Yes, it is in bulk mode.
This device is first revision so 3103b is not relevant.
m88ds3103 sets it to 16000, inside em28xx module.
With some modifications, i managed to run https://raw.githubusercontent.com/CoreELEC/media_tree_aml/master/drivers/media/platform/meson/dvb/ds3k.c
Neither of the two drivers work with ts_clk 24000

b-rad-NDi, 2022-04-05:
Is this with a specific mux or satellite configuration? No one else has complained to me of this. If we can identify the exact situation I'll try and address it without affecting anyone else.

b-rad-NDi, 2022-04-05:
There are many, many, many thousands of these devices in use in the field and no one has mentioned this to me. Is this issue new or long standing? It's confusing that it has never been brought to my attention.

psyborg55, 2022-04-05:
Assuming no one else went so deep in driver debugging? I think you can reproduce the problem even without satellite installation, i will check and update here.
I assume it is long standing since the driver was upstreamed with reduced clock of 16000

b-rad-NDi, 2022-04-05:
I would think corrupted transport stream would be reported to me, even by standard end users without debug skills, if this were systemic.

I'll investigate and produce a build with the ts clock reduced for wide testing, to ensure it doesn't affect anyone aversely.

psyborg55, 2022-04-05:
Don't get me wrong, but standard end users wont recompile driver, and go change all kinds of values in there to notice something isnt working as it should with reference driver.
The change that revealed the bug is about clock increase 16000>24000, not reduction.
There is a whole lot of fixes available if you wish to include them in the driver: https://github.com/deeptho/linux_media/pull/2

b-rad-NDi, 2022-04-05:
Standard end users will not recompile a driver, but trust me, they will complain like no other if anything is broken.

Are all of those patches written or sourced by you? If so, do I have permission to integrate and upstream them?

psyborg55, 2022-04-05:
Yes, of course. But it would be good if some users test it before submission. And some of them are related to features that are not finished yet, like automatic selection of delivery system and spectrum scan

b-rad-NDi, 2022-04-05:
They'll be integrated here and then tested as much as possible first to ensure no breakage. If there is anything from your pulll request that is not finished besides those two just point them out. Most likely automatic selection of delivery system wouldn't be accepted upstream without core changes, my dvb-t/t2 auto plp patch was not accepted for this reason.

psyborg55, 2022-04-05:
I'm pretty sure it can be done without core changes. In fact i already made it work with ds3k driver linked above, only i have to send tune request twice for lock to succeed, but it worked with both DVBS and DVBS2 transponder. m88ds3103 driver is more complex, but the principle is the same. I just need to spend a bit more time with it to make it work.
Then all the known problems and missing features will be solved, leaving only ts_clk bug (if it is an actual bug) and blindscan.
Problem with blindscan is datasheet is missing half of register descriptions, for example how to obtain SNR during FFT scan, symbol rate sweeping and constellation monitor are listed but not mentioned in the register list...
Do you have any other datasheet version? DS3000 or TS2020 might be helpful as well

b-rad-NDi, 2022-04-06:
It's not a matter of making it work, it's a matter of being accepted upstream. I received pushback for dvb-t/t2 auto and it was never accepted. I only have m88ds3103 datasheet I believe. I'm not sure if I have ts2020, I'll look around.

psyborg55, 2022-04-06:
And if the change is as simple as https://github.com/deeptho/linux_media/pull/2/commits/50864e17e99015fba5de766c4b4d91c69a38ac10 they wont accept it? Then just skip auto delivery system commits.
In fact i dont care if they wont accept anything at all, for that reason there are media trees that host updated drivers, like yours and deepthought's

psyborg55, 2022-04-11:
I tried to port ts_out routine from reference driver as well as the ts_divide routine, and include them into init and set_frontend callbacks. It made no difference at all.
Since there is no m88ds3103 device in kernel that have ts_clk > 16000 i can only assume both reference driver and datasheet are in error. Maybe a CP mistake from previous hw revisions.
There is also some sort of mclk enhance register, but changing it has zero effect on ts_clk

b-rad-NDi, 2022-05-11:
I'm still finding time to work on upstreaming this.

Would you by chance like a 461e with 3103b demod mailed to you for free?

b-rad-NDi, 2022-05-11:
Cool. Send an email with your mailing address to my open source email address (just view commits)

psyborg55, 2022-06-20:
> I'm still finding time to work on upstreaming this.
> 
> Would you by chance like a 461e with 3103b demod mailed to you for free?

you can upstream these commits if you wish (they do not depend on tbs media tree)

https://github.com/deeptho/linux_media/pull/2/commits/743dac868e30f158b853f3ecbc0c8830330dc7ae

https://github.com/deeptho/linux_media/pull/2/commits/706750f5f3e4fbebc1e4ed92b99b4f1407db24f3

https://github.com/deeptho/linux_media/pull/2/commits/e2dcd9cb329228c40b9dd32397d961a6999e6195 with https://github.com/deeptho/linux_media/pull/2/commits/38e92e53a22bb531ca54aaa52a71b44e7491d5b5

https://github.com/deeptho/linux_media/pull/2/commits/770993388185f9853712e46a55660663d4e35ca4

https://github.com/deeptho/linux_media/pull/2/commits/45b110534507ed54b901f72fb5ab057bc11196f1 with
https://github.com/deeptho/linux_media/pull/2/commits/9765629467f53d4e0e43e23af3ede3a5f810fa07

https://github.com/deeptho/linux_media/pull/2/commits/7649c8ce6586226410a04f66175ce691ce514199

https://github.com/deeptho/linux_media/pull/2/commits/95ba2d442104faf5e6845f91fea55223860353c0

https://github.com/deeptho/linux_media/pull/2/commits/6c3c8d9bd5a84deb4764661899c455c6f6cab002

https://github.com/deeptho/linux_media/pull/2/commits/eddc75a74f5ee85d618a7f7f043d001a99756b7a

https://github.com/deeptho/linux_media/pull/2/commits/cf455fc8b2ea1f5534598b727867e91da8e8ac5d

https://github.com/deeptho/linux_media/pull/2/commits/92bf381dcde33c2c6f65ff09fe9c116c6f761632

https://github.com/deeptho/linux_media/pull/2/commits/db0c725861a690b0f3c20950a147772a1c2b40cb

https://github.com/deeptho/linux_media/pull/2/commits/9d2ca13308d9206dc9b7f86c31de368bc3b74e8e with
https://github.com/deeptho/linux_media/pull/2/commits/98cdf535ef17050683df04d8a364e437735e122a

https://github.com/deeptho/linux_media/pull/2/commits/ed3442eb6f1a6c872aeebe38e4d3c327b2712bd8

psyborg55, 2022-09-29:
An update to the issue:
While doing some work on spectrum scan i noticed the device cannot get all 512 samples from ADC,  ~60 samples at the end of each 96mhz window have large amplitude and are useless. Looks like a nasty low level bug, then i remembered ts clk could go above 20000 but never found precise value and i thought maybe the problems are related.
Max ts clk at which everything is good is 21333, multiplied with 4 = 85332
Next divided that value with FFT step which is 187,5 to get 455.1 which is aproximate number of valid samples.
The actual number of valid samples is 451, the small difference is due to fact that the spectrum code rounds FFT step to 187 and ADC sample to 95744.
So the issues are indeed related.

psyborg55, 2022-09-30:
Here the screenshot of spectrum scan with full ADC window. Approximately 11.4mhz at the end of each sample are garbled. I did try to lower ts_clk back to 16mhz, even use serial mode, and reduced mclk to 144 and 96 mhz, the result is the same.
So the ts_clk is not a cause of this it just suffers from same underlying bug.
Which may be at firmware level or even in hardware
[image]