b-rad-NDi/Ubuntu-media-tree-kernel-builder #140: cx23885[1]: mpeg risc op code error

micenusa, 2022-04-01:
Hi,
I see this cx23885[1]: mpeg risc op code error on the [dmesg](https://docs.google.com/document/d/13R8_cdemMv0OXKWXQR6usYVHEIaJsmWTcQptozySp9g/edit?usp=sharing)!
How to solve it?

b-rad-NDi, 2022-04-04:
Have you read through any of the other issues describing this?

micenusa, 2022-04-05:
Hi, I'm very new to this Linux topic.
If there is already an issue with the solving patch, please close this and redirect me!
btw...In ubuntu 18 it works perfectly...Any version above returns dmesg errors!

shspvr, 2022-04-05:
> Hi, I'm very new to this Linux topic. If there is already an issue with the solving patch, please close this and redirect me! btw...In ubuntu 18 it works perfectly...Any version above returns dmesg errors!

what your system spec.

micenusa, 2024-09-22:
not important

closed 2024-09-22 by micenusa