b-rad-NDi/Ubuntu-media-tree-kernel-builder #142: Jammy build

TelepathicWalrus, 2022-04-25:
Hi,

I've recently installed my server to 22.04 from 20.04 and was wondering when you were planning on updating the repository for the new ubuntu? It would be really nice to have TV back :) 

Thanks

shspvr, 2022-04-25:
what tuner ?

b-rad-NDi, 2022-04-25:
I have a new Jammy build built, I'll push it to my testing ppa tomorrow.

TelepathicWalrus, 2022-05-07:
Hey, any update on the build push?

pkloehn41, 2022-05-11:
Interested as well and thank you for the work to support this hardware for ubuntu. My hardware is the WinTV-quadHD PCIe model.

b-rad-NDi, 2022-05-13:
If you enable my test ppa Jammy packages are up now. I haven't gotten around to testing them, but they should work.

Raithmir, 2022-05-17:
Just wanted to confirm Jammy packages appear to be working all OK.

b-rad-NDi, 2022-05-17:
Thank you, I've pushed them to the main ppa.

closed 2022-05-17 by b-rad-NDi