b-rad-NDi/Ubuntu-media-tree-kernel-builder #143: quadHD PCIe card issues

plexyTV, 2022-05-10:
Hello-
I've been running my card for a few years now on Linux Mint.  I'm currently on Mint 20.3 with 5.4.0-962201191901-generic kernel.  Neither of the more recent 5.4.0-x releases work with my card - I cannot tune any channels.  Please point me in the right direction to help you debug this.

Thanks.

Saentist, 2022-05-11:
Start with **DMESG** message's.

plexyTV, 2022-05-13:
Thanks for the quick reply.  I noticed the following in dmesg when I tried to tune a channel:

[  456.045811] si2157 5-0060: found a 'Silicon Labs Si2157-A30 ROM 0x50'
[  456.047536] si2157 5-0060: Can't continue without a firmware.

I don't recall having to explicitly install firmware before, but a simple 'sudo apt-get install linux-firmware-hauppauge' appears to have corrected my issue.  I'm now running 5.4.0-1042203222222-generic.  Go ahead and close this out.  Thanks again!