b-rad-NDi/Ubuntu-media-tree-kernel-builder #145: linux-hwe-mediatree Package for Ubuntu 20.04 does not Support recent HWE Kernel

KaeTuuN, 2022-08-02:
linux-hwe-mediatree is still at Kernel 5.13, while Ubuntu 20.04 HWE is already at 5.15.

@b-rad-NDi How long will it take to Update it? Is it otherwise possible to use the Jammy Version?

xxfantasiadownxx, 2022-08-10:
I am also having this issue with the package. This is the command and output error I am receiving:

`sudo apt-get install linux-hwe-mediatree`

Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 linux-headers-5.13.0-442206031516-generic : Depends: libssl1.1 (>= 1.1.0) but it is not installable
 linux-modules-extra-5.13.0-442206031516-generic : Depends: crda but it is not installable or
                                                            wireless-crda but it is not installable
E: Unable to correct problems, you have held broken packages.

Ubuntu 22.04.1 LTS
5.15.0-43-generic Kernel

poopingpenis, 2023-01-19:
Would also like a Jammy kernel on Raspberry Pi :)