b-rad-NDi/Ubuntu-media-tree-kernel-builder #147: quadhd signal cuts out with rain and wind.

liam8888999, 2022-08-24:
regular tv signal in actual tv still works with small artefacts but the quadhd completely cuts out often and when working is unwatchable. ubuntu 22.04

Saentist, 2022-08-27:
> regular tv signal in actual tv still works with small artefacts but the quadhd completely cuts out often and when working is unwatchable. ubuntu 22.04

Your signal is on EDGE of stability, get better antenna, or change location of antenna.

Software you use ignore broken CC segments, with stop picture.
On TV's visualization Chip do think different, and you see less artifacts.

liam8888999, 2022-08-27:
do you know if a hdhomerun acts more like a tv visualisation chip or the software for the broken cc segments

Saentist, 2022-08-27:
> do you know if a hdhomerun acts more like a tv visualisation chip or the software for the broken cc segments

act as network tuner.

liam8888999, 2022-08-28:
do you know how well it does with low signal though before I go and buy one

Saentist, 2022-08-28:
> do you know how well it does with low signal though before I go and buy one

No idea.
Is't more simple to fix antenna problem then invest to new devices.
check cable and connections, activate LNA