b-rad-NDi/Ubuntu-media-tree-kernel-builder #148: Can not install more than 4 x Hauppauge WinTV-QuadHD-DVB in a system

Znuff, 2022-09-04:
Hello,

As per title, I've been struggling to use 5 x Hauppauge WinTV-QuadHD-DVB in a single system without any success.

It seems that the driver will correctly identify 4 of the cards, but the fifth one will fail. These 

```
Linux version 5.13.0-442206031516-generic (buildd@lcy02-amd64-103) (gcc (Ubuntu 9.4.0-1ubuntu1~20.04.1) 9.4.0, GNU ld (GNU Binutils for Ubuntu) 2.34) #0+mediatree+hauppauge~hwe-Ubuntu SMP Mon Jun 6 18:58:29 UTC 202 (Ubuntu 5.13.0-442206031516.0+mediatree+hauppauge~hwe-generic 5.13.19)
cx23885: CORE cx23885[0]: subsystem: 0070:6a28, board: Hauppauge WinTV-QuadHD-DVB [card=56,autodetected]
tveeprom: Hauppauge model 166100, rev B4I6, serial# 4036162686
cx23885: cx23885[0]: hauppauge eeprom: model=166100
cx23885: CORE cx23885[1]: subsystem: 0070:6b28, board: Hauppauge WinTV-QuadHD-DVB [card=56,autodetected]
tveeprom: Hauppauge model 166101, rev B4I6, serial# 4036162686
cx23885: cx23885[1]: hauppauge eeprom: model=166101
cx23885: CORE cx23885[2]: subsystem: 0070:6a28, board: Hauppauge WinTV-QuadHD-DVB [card=56,autodetected]
tveeprom: Hauppauge model 166100, rev B4I6, serial# 4036162869
cx23885: cx23885[2]: hauppauge eeprom: model=166100
cx23885: CORE cx23885[3]: subsystem: 0070:6b28, board: Hauppauge WinTV-QuadHD-DVB [card=56,autodetected]
tveeprom: Hauppauge model 166101, rev B4I6, serial# 4036162869
cx23885: cx23885[3]: hauppauge eeprom: model=166101
cx23885: CORE cx23885[4]: subsystem: 0070:6a28, board: Hauppauge WinTV-QuadHD-DVB [card=56,autodetected]
tveeprom: Hauppauge model 166100, rev B4I6, serial# 4036152220
cx23885: cx23885[4]: hauppauge eeprom: model=166100
cx23885: CORE cx23885[5]: subsystem: 0070:6b28, board: Hauppauge WinTV-QuadHD-DVB [card=56,autodetected]
tveeprom: Hauppauge model 166101, rev B4I6, serial# 4036152220
cx23885: cx23885[5]: hauppauge eeprom: model=166101
cx23885: CORE cx23885[6]: subsystem: 0070:6a28, board: Hauppauge WinTV-QuadHD-DVB [card=56,autodetected]
tveeprom: Hauppauge model 166100, rev B4I6, serial# 4036162718
cx23885: cx23885[6]: hauppauge eeprom: model=166100
cx23885: CORE cx23885[7]: subsystem: 0070:6b28, board: Hauppauge WinTV-QuadHD-DVB [card=56,autodetected]
tveeprom: Hauppauge model 166101, rev B4I6, serial# 4036162718
cx23885: cx23885[7]: hauppauge eeprom: model=166101
cx23885: CORE cx23885[8]: subsystem: 0070:6a28, board: Hauppauge WinTV-HVR1800 [card=2,insmod option]
tveeprom: Hauppauge model 166100, rev B5I6, serial# 4036177626
cx23885: cx23885[8]: hauppauge eeprom: model=166100
```

Please note that I DID NOT pass the `card=2` option, this is somehow automatically inserted.

I have attempted to use pass `card=56,56,56,56,56,56,56,56,56,56`, but the kernel will complain with:

```
cx23885: `56' invalid for parameter `card'
card: can only take 8 arguments
```

The devices all seem identical, as per `lspci`:

```
03:00.0 Multimedia video controller: Conexant Systems, Inc. CX23887/8 PCIe Broadcast Audio and Video Decoder with 3D Comb (rev 04)
	Subsystem: Hauppauge computer works Inc. CX23887/8 PCIe Broadcast Audio and Video Decoder with 3D Comb
	Flags: bus master, fast devsel, latency 0, IRQ 96
	Memory at fc800000 (64-bit, non-prefetchable) [size=2M]
	Capabilities: [40] Express Endpoint, MSI 00
	Capabilities: [80] Power Management version 3
	Capabilities: [90] Vital Product Data
	Capabilities: [a0] MSI: Enable- Count=1/1 Maskable- 64bit+
	Capabilities: [100] Advanced Error Reporting
	Capabilities: [200] Virtual Channel
	Kernel driver in use: cx23885
	Kernel modules: cx23885

04:00.0 Multimedia video controller: Conexant Systems, Inc. CX23887/8 PCIe Broadcast Audio and Video Decoder with 3D Comb (rev 04)
	Subsystem: Hauppauge computer works Inc. CX23887/8 PCIe Broadcast Audio and Video Decoder with 3D Comb
	Flags: bus master, fast devsel, latency 0, IRQ 98
	Memory at fc600000 (64-bit, non-prefetchable) [size=2M]
	Capabilities: [40] Express Endpoint, MSI 00
	Capabilities: [80] Power Management version 3
	Capabilities: [90] Vital Product Data
	Capabilities: [a0] MSI: Enable- Count=1/1 Maskable- 64bit+
[... 190 lines cut ...]
  cx23885_initdev+0x966/0x980 [cx23885]
  cx23885_init+0x36/0x1000 [cx23885]
 UBSAN: array-index-out-of-bounds in /root/media_build/v4l/cx23885-core.c:909:20
  cx23885_initdev+0x949/0x980 [cx23885]
  cx23885_init+0x36/0x1000 [cx23885]
 UBSAN: array-index-out-of-bounds in /root/media_build/v4l/cx23885-core.c:998:2
  cx23885_initdev+0x934/0x980 [cx23885]
  cx23885_init+0x36/0x1000 [cx23885]
 cx23885: CORE cx23885[8]: subsystem: 0070:6a28, board: Hauppauge WinTV-HVR1800lp [card=1,insmod option]
 cx23885: cx23885[8]: hauppauge eeprom: model=166100
 cx25840 37-0044: cx23888 A/V decoder found @ 0x88 (cx23885[8])
 cx25840 37-0044: loaded v4l-cx23885-avcore-01.fw firmware (16382 bytes)
 cx23885: cx23885_dvb_register() allocating 1 frontend(s)
 cx23885: cx23885[8]: cx23885 based dvb card
 cx23885: cx23885[8]: frontend initialization failed
 cx23885: cx23885_dvb_register() dvb_register failed err = -22
 cx23885: cx23885_dev_setup() Failed to register dvb on VID_C
 cx23885: cx23885_dev_checkrevision() Hardware revision = 0xd0
 cx23885: cx23885[8]/0: found at 0000:15:00.0, rev: 4, irq: 33, latency: 0, mmio: 0xfb400000
 cx23885: CORE cx23885[9]: subsystem: 0070:6b28, board: UNKNOWN/GENERIC [card=0,insmod option]
 cx23885: cx23885_dev_checkrevision() Hardware revision = 0xd0
 cx23885: cx23885[9]/0: found at 0000:16:00.0, rev: 4, irq: 41, latency: 0, mmio: 0xfb200000
 ```

Please advise.

Znuff, 2022-09-04:
I'll post the resolution, as I've finally figured out what's going on.

The `cx23885` defines a maximum number of 8 devices, for some... reasons:

```
#define CX23885_MAXBOARDS 8
```

Changing this value to `16` technically fixes the driver issue. You can now pass up to 16 parameters to `card=`, or just let the module autodetect, as it will do a good job by itself.

Next thing you'll find out is that the default Ubuntu kernel defines a `CONFIG_DVB_MAX_ADAPTERS = 8`, for, again, some obscure reasons.

The author of this PPA, @b-rad-NDi patches the setting itself to `16`, as per this comment: https://github.com/b-rad-NDi/Ubuntu-media-tree-kernel-builder/issues/40#issuecomment-371512505

I attempted to use the script in this repository to plug in my own changes and make the all the debs in the PPA, but I was highly unsuccessful - there are no instructions, and I couldn't figure out the correct procedure/order of the operations.

In the end, I managed to re-compile the debs posted, by downloading the source debs of this project and following the re-build instructions after making my changes. 

I've followed this old article: https://raphaelhertzog.com/2010/12/15/howto-to-rebuild-debian-packages/

End result:

```
[    4.763345] cx23885: CORE cx23885[0]: subsystem: 0070:6a28, board: Hauppauge WinTV-QuadHD-DVB [card=56,insmod option]
[    5.119305] tveeprom: Hauppauge model 166100, rev B4I6, serial# 4036162686
[    5.119319] cx23885: cx23885[0]: hauppauge eeprom: model=166100
[    5.149307] cx23885: CORE cx23885[1]: subsystem: 0070:6b28, board: Hauppauge WinTV-QuadHD-DVB [card=56,insmod option]
[    5.507337] tveeprom: Hauppauge model 166101, rev B4I6, serial# 4036162686
[    5.507350] cx23885: cx23885[1]: hauppauge eeprom: model=166101
[    5.531103] cx23885: CORE cx23885[2]: subsystem: 0070:6a28, board: Hauppauge WinTV-QuadHD-DVB [card=56,insmod option]
[    5.884254] tveeprom: Hauppauge model 166100, rev B4I6, serial# 4036162869
[    5.884268] cx23885: cx23885[2]: hauppauge eeprom: model=166100
[    5.907618] cx23885: CORE cx23885[3]: subsystem: 0070:6b28, board: Hauppauge WinTV-QuadHD-DVB [card=56,insmod option]
[    6.260363] tveeprom: Hauppauge model 166101, rev B4I6, serial# 4036162869
[    6.260376] cx23885: cx23885[3]: hauppauge eeprom: model=166101
[    6.285376] cx23885: CORE cx23885[4]: subsystem: 0070:6a28, board: Hauppauge WinTV-QuadHD-DVB [card=56,insmod option]
[    6.636594] tveeprom: Hauppauge model 166100, rev B4I6, serial# 4036152220
[    6.636608] cx23885: cx23885[4]: hauppauge eeprom: model=166100
[    6.661288] cx23885: CORE cx23885[5]: subsystem: 0070:6b28, board: Hauppauge WinTV-QuadHD-DVB [card=56,insmod option]
[    7.012620] tveeprom: Hauppauge model 166101, rev B4I6, serial# 4036152220
[    7.012633] cx23885: cx23885[5]: hauppauge eeprom: model=166101
[    7.037313] cx23885: CORE cx23885[6]: subsystem: 0070:6a28, board: Hauppauge WinTV-QuadHD-DVB [card=56,insmod option]
[    7.388613] tveeprom: Hauppauge model 166100, rev B4I6, serial# 4036162718
[    7.388626] cx23885: cx23885[6]: hauppauge eeprom: model=166100
[    7.413091] cx23885: CORE cx23885[7]: subsystem: 0070:6b28, board: Hauppauge WinTV-QuadHD-DVB [card=56,insmod option]
[    7.764609] tveeprom: Hauppauge model 166101, rev B4I6, serial# 4036162718
[    7.764622] cx23885: cx23885[7]: hauppauge eeprom: model=166101
[    7.789148] cx23885: CORE cx23885[8]: subsystem: 0070:6a28, board: Hauppauge WinTV-QuadHD-DVB [card=56,autodetected]
[    8.140617] tveeprom: Hauppauge model 166100, rev B5I6, serial# 4036177626
[    8.140631] cx23885: cx23885[8]: hauppauge eeprom: model=166100
[    8.165655] cx23885: CORE cx23885[9]: subsystem: 0070:6b28, board: Hauppauge WinTV-QuadHD-DVB [card=56,autodetected]
[    8.516558] tveeprom: Hauppauge model 166101, rev B5I6, serial# 4036177626
[    8.516571] cx23885: cx23885[9]: hauppauge eeprom: model=166101
```

All 5 cards are detected, all 20 adapters are showing up properly.

Saentist, 2023-04-10:
UEFI + Secure boot is used.