b-rad-NDi/Ubuntu-media-tree-kernel-builder #154: No channels found on latest Ubuntu Jammy RPi drivers

poopingpenis, 2023-04-01:
I installed the latest RPi drivers for Jammy (thank you!) but when I went to set up tvheadend, it wouldn't find any channels. It wouldn't even scan for muxes, just said immediately the scan was finished with 0 muxes and 0 services found. I tried again with w_scan and got these resutls:

```
w_scan -fa -c PR 
w_scan version 20170107 (compiled for DVB API 5.11)
using settings for PUERTO RICO
Country identifier PR not defined. Using defaults.
scan type TERRCABLE_ATSC, channellist 1
output format vdr-2.0
WARNING: could not guess your codepage. Falling back to 'UTF-8'
output charset 'UTF-8', use -C <charset> to override
Info: using DVB adapter auto detection.
	/dev/dvb/adapter0/frontend0 -> TERRCABLE_ATSC "MaxLinear MxL692 VSB tuner-demodulator": good :-)
	/dev/dvb/adapter1/frontend0 -> TERRCABLE_ATSC "MaxLinear MxL692 VSB tuner-demodulator": good :-)
	/dev/dvb/adapter2/frontend0 -> TERRCABLE_ATSC "MaxLinear MxL692 VSB tuner-demodulator": good :-)
	/dev/dvb/adapter3/frontend0 -> TERRCABLE_ATSC "MaxLinear MxL692 VSB tuner-demodulator": good :-)
Using TERRCABLE_ATSC frontend (adapter /dev/dvb/adapter0/frontend0)
-_-_-_-_ Getting frontend capabilities-_-_-_-_ 
Using DVB API 5.11
frontend 'MaxLinear MxL692 VSB tuner-demodulator' supports
INVERSION_AUTO
8VSB
FREQ (54.00MHz ... 858.00MHz)
-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_ 
57000: 8VSB(time: 00:00.687) 
63000: 8VSB(time: 00:03.735) 
69000: 8VSB(time: 00:06.739) 
79000: 8VSB(time: 00:09.791) 
85000: 8VSB(time: 00:12.795) 
177000: 8VSB(time: 00:15.843) 
183000: 8VSB(time: 00:18.895) 
189000: 8VSB(time: 00:21.947) 
195000: 8VSB(time: 00:24.951) 
201000: 8VSB(time: 00:28.003) 
207000: 8VSB(time: 00:31.007) 
213000: 8VSB(time: 00:34.059) 
473000: 8VSB(time: 00:37.107) 
479000: 8VSB(time: 00:40.111) 
485000: 8VSB(time: 00:43.115) 
491000: 8VSB(time: 00:46.163) 
497000: 8VSB(time: 00:49.167) 
503000: 8VSB(time: 00:52.171) 
509000: 8VSB(time: 00:55.175) 
515000: 8VSB(time: 00:58.179) 
521000: 8VSB(time: 01:01.183) 
527000: 8VSB(time: 01:04.230) 
533000: 8VSB(time: 01:07.282) 
539000: 8VSB(time: 01:10.290) 
545000: 8VSB(time: 01:13.294) 
551000: 8VSB(time: 01:16.298) 
557000: 8VSB(time: 01:19.350) 
563000: 8VSB(time: 01:22.402) 
569000: 8VSB(time: 01:25.454) 
575000: 8VSB(time: 01:28.498) 
581000: 8VSB(time: 01:31.546) 
587000: 8VSB(time: 01:34.550) 
593000: 8VSB(time: 01:37.602) 
599000: 8VSB(time: 01:40.654) 
605000: 8VSB(time: 01:43.702) 
611000: 8VSB(time: 01:46.706) 
617000: 8VSB(time: 01:49.758) 
623000: 8VSB(time: 01:52.806) 
629000: 8VSB(time: 01:55.858) 
635000: 8VSB(time: 01:58.862) 
641000: 8VSB(time: 02:01.910) 
647000: 8VSB(time: 02:04.961) 
653000: 8VSB(time: 02:08.013) 
659000: 8VSB(time: 02:11.017) 
665000: 8VSB(time: 02:14.069) 
671000: 8VSB(time: 02:17.073) 
677000: 8VSB(time: 02:20.125) 
683000: 8VSB(time: 02:23.129) 
689000: 8VSB(time: 02:26.133) 
695000: 8VSB(time: 02:29.177) 
701000: 8VSB(time: 02:32.185) 
707000: 8VSB(time: 02:35.237) 
713000: 8VSB(time: 02:38.289) 
719000: 8VSB(time: 02:41.341) 
725000: 8VSB(time: 02:44.345) 
731000: 8VSB(time: 02:47.397) 
737000: 8VSB(time: 02:50.449) 
743000: 8VSB(time: 02:53.453) 
749000: 8VSB(time: 02:56.501) 
755000: 8VSB(time: 02:59.505) 
761000: 8VSB(time: 03:02.557) 
767000: 8VSB(time: 03:05.600) 
773000: 8VSB(time: 03:08.604) 
779000: 8VSB(time: 03:11.656) 
785000: 8VSB(time: 03:14.708) 
791000: 8VSB(time: 03:17.760) 
797000: 8VSB(time: 03:20.764) 
803000: 8VSB(time: 03:23.812) 

ERROR: Sorry - i couldn't get any working frequency/transponder
 Nothing to scan!!

```

As you can see, the tuner itself seems to be recognised without issue, but nothing is found.

Saentist, 2023-04-01:
Attach DMESG output 
`dmesg | egrep "firmware|dvb|frontend|usb"`

poopingpenis, 2023-04-01:
> Attach DMESG output

Here you go. I forgot to mention, this issue is with the WinTV-quadHD. I just tried scanning with my WinTV-dualHD and had no issues, but quadHD is still giving me the same problem.

```
$ sudo dmesg | egrep "firmware|dvb|frontend|usb"
[ 5779.205859] usb 1-1.1: USB disconnect, device number 3
[ 5779.205890] usb 1-1.1.1: USB disconnect, device number 4
[ 5779.227593] usb 1-1.1.3: USB disconnect, device number 5
[ 5836.750515] usb 1-1.1: new high-speed USB device number 6 using xhci_hcd
[ 5836.851421] usb 1-1.1: New USB device found, idVendor=2040, idProduct=826d, bcdDevice= 1.00
[ 5836.851457] usb 1-1.1: New USB device strings: Mfr=3, Product=1, SerialNumber=2
[ 5836.851473] usb 1-1.1: Product: 955D
[ 5836.851488] usb 1-1.1: Manufacturer: HCW
[ 5836.851502] usb 1-1.1: SerialNumber: 0014115250
[ 5838.202521] em28xx 1-1.1:1.0: dvb set to bulk mode.
[ 5839.547067] em28xx 1-1.1:1.0: dvb ts2 set to bulk mode.
[ 5839.804906] dvbdev: DVB: registering new adapter (1-1.1:1.0)
[ 5839.804917] em28xx 1-1.1:1.0: DVB: registering adapter 0 frontend 0 (LG Electronics LGDT3306A VSB/QAM Frontend)...
[ 5839.804930] dvbdev: dvb_create_media_entity: media entity 'LG Electronics LGDT3306A VSB/QAM Frontend' registered.
[ 5839.806149] dvbdev: dvb_create_media_entity: media entity 'dvb-demux' registered.
[ 5839.831075] dvbdev: DVB: registering new adapter (1-1.1:1.0)
[ 5839.831089] em28xx 1-1.1:1.0: DVB: registering adapter 1 frontend 0 (LG Electronics LGDT3306A VSB/QAM Frontend)...
[ 5839.831104] dvbdev: dvb_create_media_entity: media entity 'LG Electronics LGDT3306A VSB/QAM Frontend' registered.
[ 5839.832580] dvbdev: dvb_create_media_entity: media entity 'dvb-demux' registered.
[ 5839.902850] rc rc0: Hauppauge WinTV-dualHD 01595 ATSC/QAM as /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.1/1-1.1:1.0/rc/rc0
[ 5839.903087] input: Hauppauge WinTV-dualHD 01595 ATSC/QAM as /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.1/1-1.1:1.0/rc/rc0/input0
[ 5892.313264] si2157 26-0060: Using ROM firmware.
[ 5892.365540] si2157 26-0060: firmware version: 3.0.5
[ 5892.437550] si2157 27-0062: Using ROM firmware.
[ 5892.488597] si2157 27-0062: firmware version: 3.0.5
[ 6130.860415] usb 1-1.1: USB disconnect, device number 6
[ 6139.634803] usb 1-1.1: new high-speed USB device number 7 using xhci_hcd
[ 6139.735257] usb 1-1.1: New USB device found, idVendor=04e2, idProduct=0404, bcdDevice= 1.11
[ 6139.735297] usb 1-1.1: New USB device strings: Mfr=0, Product=1, SerialNumber=0
[ 6139.735314] usb 1-1.1: Product: USB 2.0 Hub
[ 6140.022810] usb 1-1.1.1: new high-speed USB device number 8 using xhci_hcd
[ 6140.123578] usb 1-1.1.1: New USB device found, idVendor=2040, idProduct=846d, bcdDevice= 1.00
[ 6140.123616] usb 1-1.1.1: New USB device strings: Mfr=3, Product=1, SerialNumber=2
[ 6140.123631] usb 1-1.1.1: Product: quadHD-A
[ 6140.123643] usb 1-1.1.1: Manufacturer: HCW
[ 6140.123654] usb 1-1.1.1: SerialNumber: 0014156253
[ 6141.490869] em28xx 1-1.1.1:1.0: dvb set to bulk mode.
[ 6142.866863] em28xx 1-1.1.1:1.0: dvb ts2 set to bulk mode.
[ 6143.075080] dvbdev: DVB: registering new adapter (1-1.1.1:1.0)
[ 6143.075092] em28xx 1-1.1.1:1.0: DVB: registering adapter 0 frontend 0 (MaxLinear MxL692 VSB tuner-demodulator)...
[ 6143.075112] dvbdev: dvb_create_media_entity: media entity 'MaxLinear MxL692 VSB tuner-demodulator' registered.
[ 6143.076358] dvbdev: dvb_create_media_entity: media entity 'dvb-demux' registered.
[ 6143.102633] dvbdev: DVB: registering new adapter (1-1.1.1:1.0)
[ 6143.102644] em28xx 1-1.1.1:1.0: DVB: registering adapter 1 frontend 0 (MaxLinear MxL692 VSB tuner-demodulator)...
[ 6143.102665] dvbdev: dvb_create_media_entity: media entity 'MaxLinear MxL692 VSB tuner-demodulator' registered.
[ 6143.104209] dvbdev: dvb_create_media_entity: media entity 'dvb-demux' registered.
[ 6143.146842] usb 1-1.1.3: new high-speed USB device number 9 using xhci_hcd
[ 6143.247665] usb 1-1.1.3: New USB device found, idVendor=2040, idProduct=846d, bcdDevice= 1.00
[ 6143.247697] usb 1-1.1.3: New USB device strings: Mfr=3, Product=1, SerialNumber=2
[ 6143.247710] usb 1-1.1.3: Product: quadHD-A
[ 6143.247720] usb 1-1.1.3: Manufacturer: HCW
[ 6143.247730] usb 1-1.1.3: SerialNumber: 0014156253
[ 6144.582900] em28xx 1-1.1.3:1.0: dvb set to bulk mode.
[ 6144.676999] usb 1-1.1: USB disconnect, device number 7
[ 6144.762899] em28xx 1-1.1.3:1.0: dvb ts2 set to bulk mode.
[ 6144.965140] usb 1-1.1.1: USB disconnect, device number 8
[ 6144.970637] dvbdev: DVB: registering new adapter (1-1.1.3:1.0)
[ 6144.970650] em28xx 1-1.1.3:1.0: DVB: registering adapter 2 frontend 0 (MaxLinear MxL692 VSB tuner-demodulator)...
[ 6144.970668] dvbdev: dvb_create_media_entity: media entity 'MaxLinear MxL692 VSB tuner-demodulator' registered.
[ 6144.972640] dvbdev: dvb_create_media_entity: media entity 'dvb-demux' registered.
[ 6144.987207] dvbdev: DVB: registering new adapter (1-1.1.3:1.0)
[ 6144.987220] em28xx 1-1.1.3:1.0: DVB: registering adapter 3 frontend 0 (MaxLinear MxL692 VSB tuner-demodulator)...
[ 6144.987241] dvbdev: dvb_create_media_entity: media entity 'MaxLinear MxL692 VSB tuner-demodulator' registered.
[ 6144.988568] dvbdev: dvb_create_media_entity: media entity 'dvb-demux' registered.
[... 77 lines cut ...]
[ 6159.062398] em28xx 1-1.1.1:1.0: DVB: registering adapter 1 frontend 0 (MaxLinear MxL692 VSB tuner-demodulator)...
[... cut ...]
[ 6162.206031] dvbdev: dvb_create_media_entity: media entity 'dvb-demux' registered.
[ 6212.920381] mxl692 23-0060: Direct firmware load for dvb-demod-mxl692.fw failed with error -2
[ 6213.059959] mxl692 25-0063: Direct firmware load for dvb-demod-mxl692.fw failed with error -2
[ 6213.192023] mxl692 27-0060: Direct firmware load for dvb-demod-mxl692.fw failed with error -2
[ 6213.332080] mxl692 29-0063: Direct firmware load for dvb-demod-mxl692.fw failed with error -2
[ 6213.473856] mxl692 23-0060: Direct firmware load for dvb-demod-mxl692.fw failed with error -2
```

Saentist, 2023-04-01:
confirm existing of 
`dvb-demod-mxl692.fw`
in 
`/lib/firmware`

poopingpenis, 2023-04-01:
> confirm existing of `dvb-demod-mxl692.fw` in `/lib/firmware`

Nope, here are the dvb* ones I have:

```
$ ls -1 /lib/firmware/dvb*
/lib/firmware/dvb-fe-xc4000-1.4.1.fw
/lib/firmware/dvb-fe-xc5000-1.6.114.fw
/lib/firmware/dvb-fe-xc5000c-4.1.30.7.fw
/lib/firmware/dvb-usb-dib0700-1.20.fw
/lib/firmware/dvb-usb-it9135-01.fw
/lib/firmware/dvb-usb-it9135-02.fw
/lib/firmware/dvb-usb-terratec-h5-drxk.fw
```

Saentist, 2023-04-01:
So install it, there is in repo firmware collection 
https://launchpad.net/~b-rad/+archive/ubuntu/kernel+mediatree+hauppauge

Then restart system or reload module's, with is faster for you ;)

poopingpenis, 2023-04-01:
The PPA doesn't have the Raspberry Pi builds, I installed it from here https://github.com/b-rad-NDi/Ubuntu-media-tree-kernel-builder/releases/tag/v0.5-RPi

Specifically, I installed all of the deb's in linux-raspi-jammy-5.15.0-arm64-23-02-28.zip and linux-raspi-jammy-tools-5.15.0-arm64-23-02-28.zip.

Saentist, 2023-04-01:
OMG too much thinking...
get deb, file extract it, use content.
Or find file online, and save it to `/lib/firmware`

poopingpenis, 2023-04-01:
That did the trick, I used the Jammy package, though (don't know if they're ultimately the same). Thanks.

closed 2023-04-01 by poopingpenis