b-rad-NDi/Ubuntu-media-tree-kernel-builder #155: 4 x Wintv Quad not detected!

estimadarocha, 2023-04-09:
Hi Brad.

Just installed a fresh new Ubuntu 22.04 and your ppa and linux-mediatree and firmware and still can't see the 16 adapters, any hint?

Thanks

Saentist, 2023-04-09:
Report result of:
`dmesg | egrep "firmware|dvb|frontend|usb"`

estimadarocha, 2023-04-09:
lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 22.04.2 LTS
Release:        22.04
Codename:       jammy

 uname -a
Linux astra 5.15.0-362206031516-generic #0+mediatree+hauppauge-Ubuntu SMP Mon Jun 6 18:58:15 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux

dmesg | egrep "firmware|dvb|frontend|usb"
[    0.318708] usbcore: registered new interface driver usbfs
[    0.318708] usbcore: registered new interface driver hub
[    0.318708] usbcore: registered new device driver usb
[    0.900378] usb usb1: New USB device found, idVendor=1d6b, idProduct=0002, bcdDevice= 5.15
[    0.900385] usb usb1: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[    0.900389] usb usb1: Product: xHCI Host Controller
[    0.900391] usb usb1: Manufacturer: Linux 5.15.0-362206031516-generic xhci-hcd
[    0.900394] usb usb1: SerialNumber: 0000:00:14.0
[    0.901835] usb usb2: New USB device found, idVendor=1d6b, idProduct=0003, bcdDevice= 5.15
[    0.901838] usb usb2: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[    0.901840] usb usb2: Product: xHCI Host Controller
[    0.901842] usb usb2: Manufacturer: Linux 5.15.0-362206031516-generic xhci-hcd
[    0.901843] usb usb2: SerialNumber: 0000:00:14.0
[    1.002887] i915 0000:00:02.0: [drm] Finished loading DMC firmware i915/kbl_dmc_ver1_04.bin (v1.4)
[    4.888874] cx23885: cx23885_dvb_register() allocating 1 frontend(s)
[    4.888875] cx23885: cx23885[0]: cx23885 based dvb card
[    4.888877] cx23885: dvb_register(): board=60 port=1
[    4.899560] si2168 5-0064: firmware version: B 4.0.2
[    4.922765] dvbdev: DVB: registering new adapter (cx23885[0])
[    4.922767] cx23885 0000:03:00.0: DVB: registering adapter 0 frontend 0 (Silicon Labs Si2168)...
[    4.922957] cx23885: cx23885_dvb_register() allocating 1 frontend(s)
[    4.922958] cx23885: cx23885[0]: cx23885 based dvb card
[    4.922959] cx23885: dvb_register(): board=60 port=2
[    4.931584] si2168 5-0066: firmware version: B 4.0.2
[    4.932821] dvbdev: DVB: registering new adapter (cx23885[0])
[    4.932822] cx23885 0000:03:00.0: DVB: registering adapter 1 frontend 0 (Silicon Labs Si2168)...
[    5.285485] cx23885: cx23885_dvb_register() allocating 1 frontend(s)
[    5.285486] cx23885: cx23885[1]: cx23885 based dvb card
[    5.285487] cx23885: dvb_register(): board=60 port=1
[    5.294698] si2168 10-0064: firmware version: B 4.0.2
[    5.295821] dvbdev: DVB: registering new adapter (cx23885[1])
[    5.295822] cx23885 0000:04:00.0: DVB: registering adapter 2 frontend 0 (Silicon Labs Si2168)...
[    5.296064] cx23885: cx23885_dvb_register() allocating 1 frontend(s)
[    5.296065] cx23885: cx23885[1]: cx23885 based dvb card
[    5.296066] cx23885: dvb_register(): board=60 port=2
[    5.305063] si2168 10-0066: firmware version: B 4.0.2
[    5.306135] dvbdev: DVB: registering new adapter (cx23885[1])
[    5.306136] cx23885 0000:04:00.0: DVB: registering adapter 3 frontend 0 (Silicon Labs Si2168)...
[    5.658099] cx23885: cx23885_dvb_register() allocating 1 frontend(s)
[    5.658103] cx23885: cx23885[2]: cx23885 based dvb card
[    5.658108] cx23885: dvb_register(): board=60 port=1
[    5.670032] si2168 15-0064: firmware version: B 4.0.2
[    5.673820] dvbdev: DVB: registering new adapter (cx23885[2])
[    5.673825] cx23885 0000:07:00.0: DVB: registering adapter 4 frontend 0 (Silicon Labs Si2168)...
[    5.674718] cx23885: cx23885_dvb_register() allocating 1 frontend(s)
[    5.674724] cx23885: cx23885[2]: cx23885 based dvb card
[    5.674729] cx23885: dvb_register(): board=60 port=2
[    5.686675] si2168 15-0066: firmware version: B 4.0.2
[    5.690439] dvbdev: DVB: registering new adapter (cx23885[2])
[    5.690444] cx23885 0000:07:00.0: DVB: registering adapter 5 frontend 0 (Silicon Labs Si2168)...
[    6.041844] cx23885: cx23885_dvb_register() allocating 1 frontend(s)
[    6.041845] cx23885: cx23885[3]: cx23885 based dvb card
[    6.041846] cx23885: dvb_register(): board=60 port=1
[    6.050971] si2168 20-0064: firmware version: B 4.0.2
[    6.052075] dvbdev: DVB: registering new adapter (cx23885[3])
[    6.052076] cx23885 0000:08:00.0: DVB: registering adapter 6 frontend 0 (Silicon Labs Si2168)...
[    6.052305] cx23885: cx23885_dvb_register() allocating 1 frontend(s)
[    6.052306] cx23885: cx23885[3]: cx23885 based dvb card
[    6.052308] cx23885: dvb_register(): board=60 port=2
[... 42 lines cut ...]
[    7.186213] cx23885: cx23885_dev_setup() Failed to register dvb adapters on VID_B
[    7.186220] cx23885: cx23885_dvb_register() allocating 1 frontend(s)
[    7.186224] cx23885: cx23885[6]: cx23885 based dvb card
[    7.186229] cx23885: dvb_register(): board=60 port=2
[    7.198220] si2168 31-0066: firmware version: B 4.0.2
[    7.201928] cx23885[6]: dvb_register_adapter failed (errno = -23)
[    7.201932] vb2_dvb_register_adapter failed (errno = -23)
[    7.202261] cx23885: cx23885_dvb_register() dvb_register failed err = -22
[    7.202275] cx23885: cx23885_dev_setup() Failed to register dvb on VID_C
[    7.553803] cx23885: cx23885_dvb_register() allocating 1 frontend(s)
[    7.553803] cx23885: cx23885[7]: cx23885 based dvb card
[    7.553804] cx23885: dvb_register(): board=60 port=1
[    7.562866] si2168 34-0064: firmware version: B 4.0.2
[    7.563935] cx23885[7]: dvb_register_adapter failed (errno = -23)
[    7.563936] vb2_dvb_register_adapter failed (errno = -23)
[    7.564039] cx23885: cx23885_dvb_register() dvb_register failed err = -22
[... cut ...]
[    7.564043] cx23885: cx23885_dvb_register() allocating 1 frontend(s)
[    7.564044] cx23885: cx23885[7]: cx23885 based dvb card
[    7.564045] cx23885: dvb_register(): board=60 port=2
[    7.573181] si2168 34-0066: firmware version: B 4.0.2
[    7.574290] cx23885[7]: dvb_register_adapter failed (errno = -23)
[    7.574291] vb2_dvb_register_adapter failed (errno = -23)
[    7.574380] cx23885: cx23885_dvb_register() dvb_register failed err = -22
[    7.574383] cx23885: cx23885_dev_setup() Failed to register dvb on VID_C

estimadarocha, 2023-04-09:
option cx23885 debug=1
option cx25840 debug=1
options tuner debug=1
options dvb-core frontend_debug=1
options dvb-core debug=1

dmesg | egrep 'si2|lgdt|cx2|dvb|tune|frontend|adapter'
[    4.462104] cx23885: cx23885 driver version 0.0.4 loaded
[    4.462133] cx23885 0000:03:00.0: enabling device (0000 -> 0002)
[    4.462241] cx23885: CORE cx23885[0]: subsystem: 0070:6a28, board: Hauppauge WinTV-QuadHD-DVB(885) [card=60,autodetected]
[    4.816739] tveeprom: tuner model is SiLabs Si2157 (idx 186, type 4)
[    4.816742] cx23885: cx23885[0]: hauppauge eeprom: model=166200
[    4.816744] cx23885: cx23885_dvb_register() allocating 1 frontend(s)
[    4.816745] cx23885: cx23885[0]: cx23885 based dvb card
[    4.816747] cx23885: dvb_register(): board=60 port=1
[    4.838100] si2168 5-0064: Silicon Labs Si2168-B40 successfully identified
[    4.838101] si2168 5-0064: firmware version: B 4.0.2
[    4.840725] si2157 6-0060: Silicon Labs Si2147/2148/2157/2158 successfully attached
[    4.840733] dvbdev: DVB: registering new adapter (cx23885[0])
[    4.840734] cx23885 0000:03:00.0: DVB: registering adapter 0 frontend 0 (Silicon Labs Si2168)...
[    4.840909] cx23885: cx23885_dvb_register() allocating 1 frontend(s)
[    4.840910] cx23885: cx23885[0]: cx23885 based dvb card
[    4.840913] cx23885: dvb_register(): board=60 port=2
[    4.849963] si2168 5-0066: Silicon Labs Si2168-B40 successfully identified
[    4.849964] si2168 5-0066: firmware version: B 4.0.2
[    4.851019] si2157 6-0062: Silicon Labs Si2147/2148/2157/2158 successfully attached
[    4.851026] dvbdev: DVB: registering new adapter (cx23885[0])
[    4.851027] cx23885 0000:03:00.0: DVB: registering adapter 1 frontend 0 (Silicon Labs Si2168)...
[    4.851245] cx23885: cx23885_dev_checkrevision() Hardware revision = 0xa5
[    4.851250] cx23885: cx23885[0]/0: found at 0000:03:00.0, rev: 4, irq: 17, latency: 0, mmio: 0x90800000
[    4.851308] cx23885 0000:04:00.0: enabling device (0000 -> 0002)
[    4.851434] cx23885: CORE cx23885[1]: subsystem: 0070:6b28, board: Hauppauge WinTV-QuadHD-DVB(885) [card=60,autodetected]
[    5.201366] tveeprom: tuner model is SiLabs Si2157 (idx 186, type 4)
[    5.201370] cx23885: cx23885[1]: hauppauge eeprom: model=166201
[    5.201371] cx23885: cx23885_dvb_register() allocating 1 frontend(s)
[    5.201372] cx23885: cx23885[1]: cx23885 based dvb card
[    5.201374] cx23885: dvb_register(): board=60 port=1
[    5.210554] si2168 10-0064: Silicon Labs Si2168-B40 successfully identified
[    5.210555] si2168 10-0064: firmware version: B 4.0.2
[    5.211613] si2157 11-0060: Silicon Labs Si2147/2148/2157/2158 successfully attached
[    5.211620] dvbdev: DVB: registering new adapter (cx23885[1])
[    5.211622] cx23885 0000:04:00.0: DVB: registering adapter 2 frontend 0 (Silicon Labs Si2168)...
[    5.212071] cx23885: cx23885_dvb_register() allocating 1 frontend(s)
[    5.212072] cx23885: cx23885[1]: cx23885 based dvb card
[    5.212073] cx23885: dvb_register(): board=60 port=2
[    5.221192] si2168 10-0066: Silicon Labs Si2168-B40 successfully identified
[    5.221193] si2168 10-0066: firmware version: B 4.0.2
[    5.222214] si2157 11-0062: Silicon Labs Si2147/2148/2157/2158 successfully attached
[    5.222220] dvbdev: DVB: registering new adapter (cx23885[1])
[    5.222221] cx23885 0000:04:00.0: DVB: registering adapter 3 frontend 0 (Silicon Labs Si2168)...
[    5.222664] cx23885: cx23885_dev_checkrevision() Hardware revision = 0xa5
[    5.222668] cx23885: cx23885[1]/0: found at 0000:04:00.0, rev: 4, irq: 18, latency: 0, mmio: 0x90600000
[    5.222710] cx23885 0000:07:00.0: enabling device (0000 -> 0002)
[    5.222824] cx23885: CORE cx23885[2]: subsystem: 0070:6a28, board: Hauppauge WinTV-QuadHD-DVB(885) [card=60,autodetected]
[    5.573917] tveeprom: tuner model is SiLabs Si2157 (idx 186, type 4)
[    5.573935] cx23885: cx23885[2]: hauppauge eeprom: model=166200
[    5.573939] cx23885: cx23885_dvb_register() allocating 1 frontend(s)
[    5.573943] cx23885: cx23885[2]: cx23885 based dvb card
[    5.573947] cx23885: dvb_register(): board=60 port=1
[    5.585959] si2168 15-0064: Silicon Labs Si2168-B40 successfully identified
[    5.585965] si2168 15-0064: firmware version: B 4.0.2
[    5.589746] si2157 16-0060: Silicon Labs Si2147/2148/2157/2158 successfully attached
[    5.589776] dvbdev: DVB: registering new adapter (cx23885[2])
[    5.589781] cx23885 0000:07:00.0: DVB: registering adapter 4 frontend 0 (Silicon Labs Si2168)...
[    5.591090] cx23885: cx23885_dvb_register() allocating 1 frontend(s)
[    5.591096] cx23885: cx23885[2]: cx23885 based dvb card
[    5.591104] cx23885: dvb_register(): board=60 port=2
[    5.603024] si2168 15-0066: Silicon Labs Si2168-B40 successfully identified
[    5.603029] si2168 15-0066: firmware version: B 4.0.2
[    5.606814] si2157 16-0062: Silicon Labs Si2147/2148/2157/2158 successfully attached
[... 104 lines cut ...]
[    7.118393] cx23885 0000:10:00.0: enabling device (0000 -> 0002)
[    7.118635] cx23885: CORE cx23885[7]: subsystem: 0070:6b28, board: Hauppauge WinTV-QuadHD-DVB(885) [card=60,autodetected]
[    7.469823] tveeprom: tuner model is SiLabs Si2157 (idx 186, type 4)
[    7.469826] cx23885: cx23885[7]: hauppauge eeprom: model=166201
[    7.469827] cx23885: cx23885_dvb_register() allocating 1 frontend(s)
[    7.469828] cx23885: cx23885[7]: cx23885 based dvb card
[    7.469829] cx23885: dvb_register(): board=60 port=1
[... cut ...]
[    7.489442] si2168 34-0066: firmware version: B 4.0.2
[    7.490495] si2157 35-0062: Silicon Labs Si2147/2148/2157/2158 successfully attached
[    7.490501] cx23885[7]: dvb_register_adapter failed (errno = -23)
[    7.490502] vb2_dvb_register_adapter failed (errno = -23)
[    7.490674] cx23885: cx23885_dvb_register() dvb_register failed err = -22
[    7.490677] cx23885: cx23885_dev_setup() Failed to register dvb on VID_C
[    7.490680] cx23885: cx23885_dev_checkrevision() Hardware revision = 0xa5

Saentist, 2023-04-09:
@b-rad-NDi 
Confirm is this a 15+ linux kernel BUG with 8 tuners limit,
and share some steps to confirm is used hauppague linux-mediatree.