b-rad-NDi/Ubuntu-media-tree-kernel-builder #161: Ubuntu 22.04.3 released with hwe kernel 6.2

cofulese, 2023-09-29:
Ubuntu 22.04.3 released with hwe kernel 6.2
Will the repo / ppa be updated for use with this release please?

charrus, 2023-11-03:
I've been building my own - got 6.5 oem working with https://github.com/b-rad-NDi/Ubuntu-media-tree-kernel-builder/issues/162#issuecomment-1793041097

Sadly with the [demise of  the linuxtv media_build backports](https://git.linuxtv.org/media_build.git/tree/README) I don't see kernels newer than [v5.18](https://git.linuxtv.org/media_build.git/tree/?h=for-v5.18) being an easy thing to publish.

johnappletree, 2023-12-17:
i am in the same boat, running 22.04 LTS with HWE enabled. Looking forward for a release that supports kernel 6.2 :-)

b-rad-NDi, 2026-02-03:
22.04 LTS HWE is not at 6.8.0

closed 2026-02-03 by b-rad-NDi