b-rad-NDi/Ubuntu-media-tree-kernel-builder #162: WinTV-Nova-S2 USB revision BAH9 support and work around

charrus, 2023-11-02:
Just repeating the issue raised here: https://github.com/b-rad-NDi/media_tree/issues/12

I created a `/etc/modprobe.d/dvb.conf` with the contents:

`options em28xx card=92`

Ran a depmod -a and rebooted.

Before:

```
[ 9204.103131] usb 1-4: new high-speed USB device number 3 using xhci_hcd
[ 9204.253860] usb 1-4: New USB device found, idVendor=2013, idProduct=0461, bcdDevice= 1.00
[ 9204.253874] usb 1-4: New USB device strings: Mfr=3, Product=1, SerialNumber=2
[ 9204.253879] usb 1-4: Product: PCTV 461
[ 9204.253884] usb 1-4: Manufacturer: PCTV
[ 9204.253887] usb 1-4: SerialNumber: 0014220822
[ 9204.295208] mc: Linux media interface: v0.10
[ 9204.306269] videodev: Linux video capture interface: v2.00
[ 9204.309508] em28xx 1-4:1.0: New device PCTV PCTV 461 @ 480 Mbps (2013:0461, interface 0, class 0)
[ 9204.309514] em28xx 1-4:1.0: DVB interface 0 found: bulk
[ 9204.367703] em28xx 1-4:1.0: chip ID is em28178
[ 9204.755107] em28xx 1-4:1.0: EEPROM ID = 26 00 01 00, EEPROM hash = 0xddd64ea2
[ 9204.755123] em28xx 1-4:1.0: EEPROM info:
[ 9204.755127] em28xx 1-4:1.0:  microcode start address = 0x0004, boot configuration = 0x01
[ 9204.764775] em28xx 1-4:1.0:  AC97 audio (5 sample rates)
[ 9204.764782] em28xx 1-4:1.0:  500mA max power
[ 9204.764786] em28xx 1-4:1.0:  Table at offset 0x27, strings=0x148c, 0x1874, 0x0a6a
[ 9204.823079] em28xx 1-4:1.0: Identified as PCTV DVB-S2 Stick (461e v2) (card=104)
[ 9204.823095] em28xx 1-4:1.0: dvb set to bulk mode.
[ 9204.823513] usbcore: registered new interface driver em28xx
[ 9204.841314] em28xx 1-4:1.0: Binding DVB extension
[ 9204.855240] em28xx: Registered (Em28xx dvb Extension) extension
[ 9204.865093] em28xx 1-4:1.0: Registering input extension
[ 9204.903053] Registered IR keymap rc-pinnacle-pctv-hd
[ 9204.904789] rc rc0: PCTV DVB-S2 Stick (461e v2) as /devices/pci0000:00/0000:00:02.5/0000:06:00.0/usb1/1-4/1-4:1.0/rc/rc0
[ 9204.904921] rc rc0: lirc_dev: driver em28xx registered at minor = 0, scancode receiver, no transmitter
[ 9204.905000] input: PCTV DVB-S2 Stick (461e v2) as /devices/pci0000:00/0000:00:02.5/0000:06:00.0/usb1/1-4/1-4:1.0/rc/rc0/input6
[ 9204.905161] em28xx 1-4:1.0: Input extension successfully initialized
[ 9204.905166] em28xx: Registered (Em28xx Input Extension) extension
```

And after:

```
[ 9284.065833] usb 1-4: new high-speed USB device number 4 using xhci_hcd
[ 9284.218453] usb 1-4: New USB device found, idVendor=2013, idProduct=0461, bcdDevice= 1.00
[ 9284.218468] usb 1-4: New USB device strings: Mfr=3, Product=1, SerialNumber=2
[ 9284.218474] usb 1-4: Product: PCTV 461
[ 9284.218479] usb 1-4: Manufacturer: PCTV
[ 9284.218483] usb 1-4: SerialNumber: 0014220822
[ 9284.244956] em28xx 1-4:1.0: New device PCTV PCTV 461 @ 480 Mbps (2013:0461, interface 0, class 0)
[ 9284.244961] em28xx 1-4:1.0: DVB interface 0 found: bulk
[ 9284.302230] em28xx 1-4:1.0: chip ID is em28178
[ 9284.689661] em28xx 1-4:1.0: EEPROM ID = 26 00 01 00, EEPROM hash = 0xddd64ea2
[ 9284.689676] em28xx 1-4:1.0: EEPROM info:
[ 9284.689681] em28xx 1-4:1.0:  microcode start address = 0x0004, boot configuration = 0x01
[ 9284.699524] em28xx 1-4:1.0:  AC97 audio (5 sample rates)
[ 9284.699533] em28xx 1-4:1.0:  500mA max power
[ 9284.699536] em28xx 1-4:1.0:  Table at offset 0x27, strings=0x148c, 0x1874, 0x0a6a
[ 9284.757817] em28xx 1-4:1.0: Identified as PCTV DVB-S2 Stick (461e) (card=92)
[ 9284.757832] em28xx 1-4:1.0: dvb set to bulk mode.
[ 9284.758086] usbcore: registered new interface driver em28xx
[ 9284.766117] em28xx 1-4:1.0: Binding DVB extension
[ 9284.777366] i2c i2c-2: Added multiplexed i2c bus 3
[ 9284.832607] ts2020 3-0060: Montage Technology TS2022 successfully identified
[ 9284.842095] a8293 2-0008: Allegro A8293 SEC successfully attached
[ 9284.842135] dvbdev: DVB: registering new adapter (1-4:1.0)
[ 9284.842142] em28xx 1-4:1.0: DVB: registering adapter 0 frontend 0 (Montage Technology M88DS3103)...
[ 9284.842154] dvbdev: dvb_create_media_entity: media entity 'Montage Technology M88DS3103' registered.
[ 9284.842869] dvbdev: dvb_create_media_entity: media entity 'dvb-demux' registered.
[ 9284.844758] em28xx 1-4:1.0: DVB extension successfully initialized
[ 9284.844765] em28xx: Registered (Em28xx dvb Extension) extension
[ 9284.852216] em28xx 1-4:1.0: Registering input extension
[ 9284.852969] Registered IR keymap rc-pinnacle-pctv-hd
[ 9284.853753] rc rc0: PCTV DVB-S2 Stick (461e) as /devices/pci0000:00/0000:00:02.5/0000:06:00.0/usb1/1-4/1-4:1.0/rc/rc0
[ 9284.853856] rc rc0: lirc_dev: driver em28xx registered at minor = 0, scancode receiver, no transmitter
[ 9284.853935] input: PCTV DVB-S2 Stick (461e) as /devices/pci0000:00/0000:00:02.5/0000:06:00.0/usb1/1-4/1-4:1.0/rc/rc0/input7
[ 9284.854102] em28xx 1-4:1.0: Input extension successfully initialized
[ 9284.854109] em28xx: Registered (Em28xx Input Extension) extension
```

And all the usual apps (like tvheadend) work perfectly.

As I have a DVB-T2 quad tuner with the same em28xx driver being used - overriding the card means only one can work at a time - so the best fix would be in the driver so card 92 is used instead of card 104.

I'm creating an issue in this repo as it looks like issues here are more noticed, and perhaps help others with a similar issue.

Sadly the media_treew backports for linuxtv.org which this repo depends on is no longer supported: https://git.linuxtv.org/media_build.git/tree/README so it'll need a fix in the kernel repo.

charrus, 2023-11-03:
Did some more digging and this was introduced into the kernel by @b-rad-NDi when adding 461e v2 support: https://github.com/torvalds/linux/commit/985b0edefa3146a59005832c2b271f9290635b31

I think the fix would simply to change https://github.com/torvalds/linux/blob/master/drivers/media/usb/em28xx/em28xx-cards.c#L2778-L2779

From:

```
	{ USB_DEVICE(0x2013, 0x0461),
			.driver_info = EM28178_BOARD_PCTV_461E_V2 },
```

To:

```
	{ USB_DEVICE(0x2013, 0x0461),
			.driver_info = EM28178_BOARD_PCTV_461E },
```

Assuming the usb id of 2013:0461 is universally a e461 and not a e461 v2 device.

charrus, 2023-11-03:
Confirmed this worked - I re-built the Ubuntu 6.5.0-1006-oem kernel packages for 22.04 with the above patch, and also changed the MAX_DVB_TUNERS to 32.

Now have 2 x WinTV-quadHD DVB-T2 PCIe cards (with em28xx driver) and a WinTV-NOVA-S2 (Rev: BAH9) with the em28xx driver all working at the same time:

```
root@vdr:~# uname -a
Linux vdr 6.5.0-1006-oem #6ubuntu1 SMP PREEMPT_DYNAMIC Fri Nov  3 19:07:36 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux
root@vdr:~# ls /dev/dvb
adapter0  adapter1  adapter2  adapter3  adapter4  adapter5  adapter6  adapter7  adapter8
```

Reception on all devices is excellent.

BerniPi, 2024-02-10:
Hello!

I have tried the workaround by setting "options em28xx card=92" and the card changes to "461e" without v2, but the card isnt available anymore in tvheadend.
Do you have any hint?