b-rad-NDi/Ubuntu-media-tree-kernel-builder #163: Attribute Error When Installing PPA

Lazer-Khat, 2023-11-02:
I am trying to install my WinTV-dualHD on my Pi (Linux CasaOS 6.1.0-rpi4-rpi-v8 #1 SMP PREEMPT Debian 1:6.1.54-1+rpt2 (2023-10-05) aarch64) to connect with my Plex Server. Whenver I run the command "sudo add-apt-repository ppa:b-rad/kernel+mediatree+hauppauge" I get the following output:

Traceback (most recent call last):
  File "/usr/bin/add-apt-repository", line 362, in <module>
    sys.exit(0 if addaptrepo.main() else 1)
                  ^^^^^^^^^^^^^^^^^
  File "/usr/bin/add-apt-repository", line 345, in main
    shortcut = handler(source, **shortcut_params)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/softwareproperties/shortcuts.py", line 40, in shortcut_handler
    return handler(shortcut, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/softwareproperties/ppa.py", line 86, in __init__
    if self.lpppa.publish_debug_symbols:
       ^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/softwareproperties/ppa.py", line 126, in lpppa
    self._lpppa = self.lpteam.getPPAByName(name=self.ppaname)
                  ^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/softwareproperties/ppa.py", line 113, in lpteam
    self._lpteam = self.lp.people(self.teamname)
                   ^^^^^^^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'people'

charrus, 2023-11-03:
Have you tried installing [python3-launchpadlib](https://help.launchpad.net/API/launchpadlib) - found in https://askubuntu.com/a/1481830

b-rad-NDi, 2026-02-03:
There is no PPA for RPi builds. You must download them directly from the releases section here. I will be pushing new RPi builds very soon.