b-rad-NDi/Ubuntu-media-tree-kernel-builder #165: Debian Instructions?

deskruf, 2024-01-12:
Just bought a [Hauppauge WinTV-dualHD](https://www.hauppauge.co.uk/site/products/data_dualhd.html) and unable to get Plex to pick up any HD channels, but it picks up all SD channels.

I came across this repo after reading some forum posts suggesting that the installation of this repo may fix the issue.

I appreciate 'Ubuntu' in the name but wondering if anyone has been able to get this installed on Debian? If so, would you mind sharing instructions?

Thought it would post here just case anyone can help before I return.

Thanks in advance!

lane-ftw, 2024-03-17:
As far as I can tell this is more or less unsupported anymore, bought mine a few years ago and have had nothing but problems. 
https://github.com/b-rad-NDi/Ubuntu-media-tree-kernel-builder/issues/61
TBH I'd return it, shame, as Hauppage! seemed to be the only decent provider besides HDHomerun.

zeufparis, 2025-06-28:
Hello Deskruf... How did you succeed in installing your WinTV-dualHD on Debian ?
My winTV-dualHD is installed (si2168 with lusb) and recognize with Kaffeine (2 tuners) but Kaffeine doesn't get any channel at all..