b-rad-NDi/Ubuntu-media-tree-kernel-builder #166: Ubuntu 24.04 Noble Numbat

betzburger, 2024-05-08:
Good evening,

any instructions available how to handle the Noble Numbat?

betzburger, 2024-06-07:
OK, now works with Noble Numbat. #116 can be closed.

Oh, one thing to add: 

There is no install for "linux-hwe-mediatree" version available. The standard "linux-mediatree" version works fine.

soba-noodle, 2024-08-19:
Confirm that linux-mediatree works with Ubuntu 24.04 LTS and WinTV quad USB on 8/18/2024.  Unfortunately the display resolution is stuck at FHD after adding mediatree.  Is there anyway to get the higher resolution?   Without mediatree, I was able to set up to 4K resolution (UHD).

betzburger, 2024-08-24:
One strange issue here. It works but for whatever reason, it is not as sensitive as on my Ubuntu 22.04. I have some stations with weaker signal. 22.04 receives them with no problem. 24.04 can't receive them. Stronger stations, no problem.

geeksmith, 2024-09-24:
@betzburger Which card are you using? No luck for me using the WinTV-HVR-955Q unless I'm using kernel 5.13 or older.

betzburger, 2024-09-25:
If have the WinTV SoloHD USB Stick

harmlessdrudge, 2024-09-26:
I have a WinTV-Nova-S2 USB tuner being identified as a PCTV Systems PCTV 461 under both LibreElec 12.0.1 and Ubuntu 24.04.1. Is this to be expected?

There is no install for "linux-hwe-mediatree" version available. The standard "linux-mediatree" version doesn't change the identification and TVHeadend 4.3 doesn't find the tuner. 

I have been using a PCI version but wish to use a mini-PC, hence the new tuner. What should I do -- wait for an update or try Ubuntu 22.04? Something else?

Martii, 2025-05-14:
Well the above news is unfortunate. 

Wasted several hours upgrading just to find out the HWE has been absent for quite some time in Noble.

Also was forced to install the `linux-mediatree` but it has some pretty serious issues with the OTA ATSC/EIT EPG modules tanking tvheadend and making it mostly unusable when enabled with a Hauppauge WinTV-quadHD PCIe.

Any updates on this? @b-rad-NDi anyone?