b-rad-NDi/Ubuntu-media-tree-kernel-builder #170: Missing support for Pinnacle 461e Rev 3 BBH9

steven-ellis, 2024-09-25:
This appears to use a newer Montage M88DS3103C demodulator 

USB details
`Bus 003 Device 007: ID 2013:0462 PCTV Systems PCTV 461`

Inserting device under Fedora 40
```
[251417.581139] usb 3-7: new high-speed USB device number 7 using xhci_hcd
[251417.711990] usb 3-7: New USB device found, idVendor=2013, idProduct=0462, bcdDevice= 1.00
[251417.712001] usb 3-7: New USB device strings: Mfr=3, Product=1, SerialNumber=2
[251417.712005] usb 3-7: Product: PCTV 461
[251417.712008] usb 3-7: Manufacturer: PCTV
[251417.712011] usb 3-7: SerialNumber: XXXXXXXXX
[251618.148955] usbcore: deregistering interface driver em28xx
```

steven-ellis, 2024-09-25:
No v3 support in

- https://github.com/torvalds/linux/blob/master/drivers/media/usb/em28xx/em28xx.h
- https://github.com/torvalds/linux/blob/master/drivers/media/dvb-frontends/m88ds3103.h

patduckless, 2024-10-02:
I have the same device. I would like to see it included in the supported hardware.

hugoboss00, 2024-10-08:
I have the same issue here. Is there a chance to include the support of 461e V3 Hardware?

psstoyanov, 2024-10-09:
Same issue for me - I've bought the device specifically for it's Linux support but it doesn't work.

I didn't expect a revision to cause scenes 😕

theMarix, 2024-10-24:
Ironically they are selling that same device now as both the PCTV 461e and the WinTV-Nova-S2. The packaging is different but it is the identical USB ID and I had the bad luck of getting the latest revision on both. :(

caramorsimon, 2024-12-19:
Ah, I have also fallen into the same trap everyone else, receiving a wintv-nova-s2 that identifies itself as a 462 in lsusb. 
Not sure how the v3 aspect of the demod was determined from the device .. .still Is there a possibility of hauppauge adding support for this version of the demod to the driver?

kolkifin, 2025-02-15:
It seems to be supported in the latest patch (https://launchpad.net/~b-rad/+archive/ubuntu/kernel+mediatree+hauppauge/+sourcepub/16956121/+listing-archive-extra)

-em28xx-Add-Hauppauge-461e-v3.patch

emveepee, 2025-02-18:
If you read the b-rad forum on GitHub you will see that Hauppauge

W11T, 2025-06-18:
> It seems to be supported in the latest patch (https://launchpad.net/~b-rad/+archive/ubuntu/kernel+mediatree+hauppauge/+sourcepub/16956121/+listing-archive-extra)
> 
> -em28xx-Add-Hauppauge-461e-v3.patch

Can you please give some details on how to apply the patch?

kolkifin, 2025-06-19:
On this page you will find instructions: [https://launchpad.net/~b-rad/+archive/ubuntu/test-test-hw-staging](url).

Micha-123, 2025-07-25:
i have compiled the Raspberrypi kernel 6.12 with the patch from here 

https://launchpad.net/~b-rad/+archive/ubuntu/test-test-hw-staging/+sourcepub/16955432/+listing-archive-extra

now i have 

`[    4.741245] em28xx 1-1.4:1.0: New device PCTV PCTV 461 @ 480 Mbps (2013:0462,                                                                                                                                                 interface 0, class 0)
[    4.741273] em28xx 1-1.4:1.0: DVB interface 0 found: bulk
[    4.798971] em28xx 1-1.4:1.0: chip ID is em28178
[    5.173807] em28xx 1-1.4:1.0: EEPROM ID = 26 00 01 00, EEPROM hash = 0x16d648                                                                                                                                                a2
[    5.173827] em28xx 1-1.4:1.0: EEPROM info:
[    5.173831] em28xx 1-1.4:1.0:        microcode start address = 0x0004, boot c                                                                                                                                                onfiguration = 0x01
[    5.182608] em28xx 1-1.4:1.0:        AC97 audio (5 sample rates)
[    5.182626] em28xx 1-1.4:1.0:        500mA max power
[    5.182629] em28xx 1-1.4:1.0:        Table at offset 0x27, strings=0x148c, 0x                                                                                                                                                1874, 0x0a6a
[    5.238845] em28xx 1-1.4:1.0: Identified as PCTV DVB-S2 Stick (461e) (card=92                                                                                                                                                )
[    5.238869] em28xx 1-1.4:1.0: dvb set to bulk mode.
[    5.239222] usbcore: registered new interface driver em28xx
[    5.260300] em28xx_dvb: loading out-of-tree module taints kernel.
[    5.260831] em28xx 1-1.4:1.0: Binding DVB extension
[    5.272979] em28xx: Registered (Em28xx dvb Extension) extension
[    5.285114] em28xx 1-1.4:1.0: Registering input extension
[    5.285737] rc rc2: lirc_dev: driver em28xx registered at minor = 0, scancode                                                                                                                                                 receiver, no transmitter
[    5.286924] em28xx 1-1.4:1.0: Input extension successfully initialized
[    5.286945] em28xx: Registered (Em28xx Input Extension) extension
`

but the demodulator on the pi wil not download firmware 

an i2c device wil be made after plug the card in 

ls /sys/bus/i2c/devices/23-0068/
modalias  name  power  subsystem  uevent

but i dont know further how to get it to work, the driver is now working, but i don know why the demodulator doesnt try to download firmware 

i know i have kernel 6.12 and this diff is from 6.11 but it has compiled without any errors on the pi kernel

MimbeF, 2025-07-27:
I am almost at the same point, but the firmware doesn't load. Which files did you use and which steps did you take, compiling the kernel and diffs, @Micha-123?

b-rad-NDi, 2026-02-03:
Please try a recent build and reopen if there is an issue.