b-rad-NDi/Ubuntu-media-tree-kernel-builder #176: PCTV 461e USB with new USB id 2013:0462 not working (raspberry pi)

deirdreobyrne, 2025-03-21:
I have a Hauppauge PCTV 461e DVB-S2 USB stick with the USB id 2013:0462. I've installed Ubuntu Jammy on a Raspberry PI 4, followed the instructions on https://github.com/b-rad-NDi/Ubuntu-media-tree-kernel-builder/releases/tag/v0.5-RPi, and the drivers aren't being loaded. The stick is visible in lsusb, but no drivers are loaded.

Micha-123, 2025-07-25:
After days of trying and compiling kernels have it get to work with the raspberrypi 4 , Raspberry OS (debian fork) and 6.12 kernel

``
29.294940] usb 1-1.4: new high-speed USB device number 7 using xhci_hcd
[ 29.383904] usb 1-1.4: New USB device found, idVendor=2013, idProduct=0462, bcdDevice= 1.00
[ 29.383942] usb 1-1.4: New USB device strings: Mfr=3, Product=1, SerialNumber=2
[ 29.383947] usb 1-1.4: Product: PCTV 461
[ 29.383953] usb 1-1.4: Manufacturer: PCTV
[ 29.383956] usb 1-1.4: SerialNumber: 0014237207
[ 29.598461] em28xx 1-1.4:1.0: New device PCTV PCTV 461 @ 480 Mbps (2013:0462, interface 0, class 0)
[ 29.598484] em28xx 1-1.4:1.0: DVB interface 0 found: bulk
[ 29.667091] em28xx 1-1.4:1.0: chip ID is em28178
[ 29.763270] usb usb2-port1: over-current change https://github.com/b-rad-NDi/Ubuntu-media-tree-kernel-builder/pull/1
[ 29.971056] usb usb2-port2: over-current change https://github.com/b-rad-NDi/Ubuntu-media-tree-kernel-builder/pull/1
[ 30.043239] em28xx 1-1.4:1.0: EEPROM ID = 26 00 01 00, EEPROM hash = 0x16d648a2
[ 30.043259] em28xx 1-1.4:1.0: EEPROM info:
[ 30.043263] em28xx 1-1.4:1.0: microcode start address = 0x0004, boot configuration = 0x01
[ 30.051621] em28xx 1-1.4:1.0: AC97 audio (5 sample rates)
[ 30.051638] em28xx 1-1.4:1.0: 500mA max power
[ 30.051641] em28xx 1-1.4:1.0: Table at offset 0x27, strings=0x148c, 0x1874, 0x0a6a
[ 30.106971] em28xx 1-1.4:1.0: Identified as PCTV DVB-S2 Stick (461e) (card=92)
[ 30.106993] em28xx 1-1.4:1.0: dvb set to bulk mode.
[ 30.107396] usbcore: registered new interface driver em28xx
[ 30.133963] em28xx_dvb: loading out-of-tree module taints kernel.
[ 30.134621] em28xx 1-1.4:1.0: Binding DVB extension
[ 30.156919] i2c i2c-23: Added multiplexed i2c bus 24
[ 30.183044] usb usb2-port3: over-current change https://github.com/b-rad-NDi/Ubuntu-media-tree-kernel-builder/pull/1
[ 30.222087] ts2020 24-0060: Montage Technology TS2022 successfully identified
[ 30.230364] a8293 23-0008: Allegro A8293 SEC successfully attached
[ 30.230461] dvbdev: DVB: registering new adapter (1-1.4:1.0)
[ 30.230470] em28xx 1-1.4:1.0: DVB: registering adapter 0 frontend 0 (Montage Technology M88DS3103C)...
[ 30.230488] dvbdev: dvb_create_media_entity: media entity 'Montage Technology M88DS3103C' registered.
[ 30.232292] dvbdev: dvb_create_media_entity: media entity 'dvb-demux' registered.
[ 30.234312] em28xx 1-1.4:1.0: DVB extension successfully initialized
[ 30.234331] em28xx: Registered (Em28xx dvb Extension) extension
[ 30.258186] em28xx 1-1.4:1.0: Registering input extension
[ 30.258465] Registered IR keymap rc-pinnacle-pctv-hd
[ 30.258820] rc rc2: PCTV DVB-S2 Stick (461e) as /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.4/1-1.4:1.0/rc/rc2
[ 30.259104] rc rc2: lirc_dev: driver em28xx registered at minor = 0, scancode receiver, no transmitter
[ 30.259202] input: PCTV DVB-S2 Stick (461e) as /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.4/1-1.4:1.0/rc/rc2/input4
[ 30.260284] em28xx 1-1.4:1.0: Input extension successfully initialized
[ 30.260298] em28xx: Registered (Em28xx Input Extension) extension
[ 30.363922] m88ds3103 23-006a: found a 'Montage Technology M88DS3103C' in cold state
[ 30.365148] m88ds3103 23-006a: downloading firmware from file 'dvb-demod-m88ds3103c.fw'
[ 30.391037] usb usb2-port4: over-current change https://github.com/b-rad-NDi/Ubuntu-media-tree-kernel-builder/pull/1
[ 30.599057] usb usb2-port1: over-current change https://github.com/b-rad-NDi/Ubuntu-media-tree-kernel-builder/pull/2
[ 30.879340] m88ds3103 23-006a: found a 'Montage Technology M88DS3103C' in warm state
[ 30.879360] m88ds3103 23-006a: firmware version: 3.8
[ 30.895005] em28xx 1-1.4:1.0: DVB: adapter 0 frontend 0 frequency 0 out of range (950000..2150000)
[ 36.116637] usb 1-1.1.1: USB disconnect, device number 5
[ 36.434909] usb 1-1.1.1: new full-speed USB device number 8 using xhci_hcd
[ 36.649801] usb 1-1.1.1: New USB device found, idVendor=0658, idProduct=0200, bcdDevice= 0.00
[ 36.649820] usb 1-1.1.1: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[ 36.657268] cdc_acm 1-1.1.1:1.0: ttyACM0: USB ACM device

``

i have changed in the em28xx-dvb.c the recognization of the version vX

So no version Check anymore, the M88DS3103C is now always load (so i think the older ones wil not qork anymore but i dont have them.

All is working fine in THV on the Rpi4, no issues.

But it was really hard to get it to work with this diff (which is made for the 6.11 ubuntu kernel)

https://launchpad.net/~b-rad/+archive/ubuntu/test-test-hw-staging/+sourcepub/16955432/+listing-archive-extra

vlapla, 2025-11-02:
Hi @Micha-123 
it's cool to read you succeeded with this DVB-S stick! I tried getting the patches from Brad Love into Debian's 6.12 kernel. First I actually just replaced drivers/media/usb/em28xx with the patched version. As a result it is now loading this driver:

Nov 02 18:38:26 raven kernel: rc rc0: PCTV DVB-S2 Stick (461e v3) as /devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3:1.0/rc/rc0
Nov 02 18:38:26 raven kernel: rc rc0: lirc_dev: driver em28xx registered at minor = 0, scancode receiver, no transmitter
Nov 02 18:38:26 raven kernel: input: PCTV DVB-S2 Stick (461e v3) as /devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3:1.0/rc/rc0/input19
Nov 02 18:38:26 raven kernel: em28xx 2-3:1.0: Input extension successfully initialized
Nov 02 18:38:26 raven kernel: em28xx: Registered (Em28xx Input Extension) extension

But in my system m88ds3103 does not appear in the system logs, and firmware is not requested (or loaded) either.

So what exactly did you change in em28xx-dvb.c? And maybe other parts of the patched version should be used? The whole drivers/media directory?

vlapla, 2025-11-07:
> If you are using the 6.12 kernel on your Pi i can send you the drivers compiled.
> 
> But you need to bump the PIs kernel to 6.12

That's very nice of you! Of course, I'd rather return to 6.12 anyway.

But on what base did you start your kernel modification? Using the kernel from https://github.com/raspberrypi/linux ?

Or did you just install linux-source from Raspberry Pi OS? That is at version 6.12.47 now (and was at 6.12.34-1+rpt1 before, according to the changelog).

Could you provide a diff between the state you started from, and what you're running now?

Sorry for the many questions!

vlapla, 2025-11-08:
Yes, I copied all modules to where the original modules from my kernel where. I was assuming autoload, like with my Ubuntu kernel, but your idea with modprobe was good!

My modules where xz compressed, so applied that compression to yours too. But now modprobe showed me this error:

`modprobe: ERROR: could not insert 'em28xx': Exec format error`

So I decompressed the modules again, now yielding a different error:

`modprobe: ERROR: could not insert 'em28xx': Unknown symbol in module, or unknown parameter (see dmesg)`

This is in dmesg:

`[ 8244.146149] em28xx: disagrees about version of symbol module_layout`

So apparently the two kernel versions are too far apart. I will try with your exact version, but not today anymore...

vlapla, 2025-11-10:
Hi @Micha-123,

so now I have a 64bit kernel with version 6.12.40-v8+ running on the Raspberry Pi 4, cross-compiled from a AMD64 system.

Still I can't insert your module:

modprobe: ERROR: could not insert 'em28xx': Exec format error

This is what dmesg shows now:

[  147.206153] em28xx: disagrees about version of symbol module_layout

I found an [answer on StackOverflow](https://stackoverflow.com/a/2722727) indicating that even a slight difference in configuration can cause this mismatch. Here is your module's module_layout:

0x39ff040a	module_layout

And this is what my other modules show:

0x6a148dd4	module_layout

So can you maybe provide your source files of the six binary modules you already uploaded? No need to make a proper patch, I suppose...

edisionnano, 2025-12-11:
How can I get the 0462 version working on latest raspberry Pi os?

edisionnano, 2025-12-11:
@Micha-123 Would it be possible for you to share the kernel?

vlapla, 2026-02-01:
Hi @edisionnano,
your patch works well for me, together with the firmware from Hauppauge (see my post here from Nov 16)! As for the Linux source, I simply installed the package linux-source (linux-source-6.12), then patched that ("patch -p1 < 462.diff" in the source directory).

vlapla, 2026-10-23:
@mifritscher2
It looks like the patch for this device was [included in Linux 7.1](https://www.opennet.ru/kernel/7.1.html) from June this year (under the title [media: em28xx: Add Hauppauge 461e v3](https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/?id=5a8db3f8f4b67c65393747179ad831d202a438e3)). According to [Linux kernel version history](https://en.wikipedia.org/wiki/Linux_kernel_version_history) (Wikipdia) the current kernel 7.3 may well become an LTS version. I guess that means it could end up in Raspberry PI OS, including the support for this TV stick. My Raspi is still running my patched Linux 6.12 (from 2024), although I see Linux 6.18 is already available in Raspberry OS.

mifritscher2, 2026-10-24:
@vlapla : Wow, thanks for the headup!

Trixie backports has already incorporated the 7.2 kernel ( https://tracker.debian.org/pkg/linux ), so "normal" Debian is covered as well.